Jieli-Tech/fw-AC63_BT_SDK #17: 关于各种OTA方式所需容量

main010, 2020-10-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

main010, 2020-10-27:
|单位(Byte)|测试盒edr|测试盒ble|BLE OTA|串口|U盘|SD卡|
|---|---|---|---|---|---|---|
|bd29| 31496  | 39228 | 59180 | x | x | x |
|br23| 29698  | 36242 | 52744 | 13368 | 16863 | 16555 |
|br25| 29498  | 36253 | 52890 | 12436 | 16100 | 15903 |

jamin3702, 2021-10-20:
[image]
编译后会提示，代码空间剩余多少，是否支持升级