Jieli-Tech/fw-AC63_BT_SDK #18: 关于芯片烧录电压

Tsai1005, 2020-10-30:
注意所选封装规格书是否VDDIO & VBAT 短接，
若是，则需要按照规格书VDDIO要求，耐压不能超过3.6V
若不是，则按照规格书VDDIO要求，耐压可支持5.5V

Williamzhangtw, 2020-10-30:
若是，    则按照规格书VDDIO要求，耐压不能超过3.6V
若不是，则按照规格书VDDIO要求，耐压可支持5.5V

??? bug?

Tsai1005, 2020-11-04:
大部分客户忽略了规格书要求，这里补充提醒

rixin-Tech, 2020-12-29:
样机芯片按VBAT和VDDIO引脚短接设计或出现以下报错(vddio 错误)，请注意烧写时USB端电压需转换！！！
[image]
使用一拖八烧写器烧写芯片VBAT和VDDIO引脚短接样机,，一拖八烧写器USB端电源电压需要转换为3.3V才能进行烧写（V2.0以上版本一拖八烧写器USB端电压PC部署工具3V~6V可设，不需要转换），否则可能出现烧录异常或者损坏芯片。USB端电压转换及接法如下：
[image]
[image]