Jieli-Tech/fw-AC63_BT_SDK #20: AC6311 SDK0.3.0  使用相关问题

tylerYu123, 2020-11-16:
1.  isd_config.ini定义了配置数据项。 在编译完成后， 如何查看分配给配置数据项的地址? 
        比如BTIF_ADR ， VM_ADR ， USERIF_ADR 的分配值 ？ 
        
    2.  isd_config.ini仅由isd_download.exe使用吗 ？     
    
    3.  isd_download.exe工具生成jl_isd.bin ，由uboot+app+cfg_tool这3文件组成， 有加密处理？
        cfg_tool的内容是如何设定， 与isd_config.ini定义的配置数据项有何关联 ？ 
        
    4.  如何查看HEAP剩余空间， 调用xPortGetFreeHeapSize() ？ 
            
    5.  如何查看任务的堆栈使用情况， 是否有溢出？
    
    6.  如何查看系统所有任务的运行状态（ 任务名，任务堆栈大小， 当前状态等） ？  
    
    7.  如何查看SDK版本号？

Tsai1005, 2020-11-16:
- Q4：调用`mem_stats` 查看内存使用情况

- Q5：任务堆栈有溢出保护，溢出会触发异常；

- Q6 ：调用`os_current_task` 查看当前任务；

rixin-Tech, 2020-11-16:
Q1：请查阅[ISD_CONFIG.INI配置文件说明](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/blob/master/doc/stuff/ISD_CONFIG.INI%E9%85%8D%E7%BD%AE%E6%96%87%E4%BB%B6%E8%AF%B4%E6%98%8E.pdf)
Q2：是的
Q3：vm、自定义数据区不加密，其他区域使用客户的秘钥加密