Jieli-Tech/fw-AC63_BT_SDK #28: 关于637N的定时中断与PWM

maxbang5, 2020-12-02:
用’AC630N_bt_data_transfer_sdk_release_v0.5.0->spp_and_le->br30‘这个SDK，AC6373B的芯片，用原先AC630N的方式写定时中断和PWM，发现用不了，代码移植烧录到AC6318上是没问题的，这两个是有什么差别么，能提供一个例程么

maxbang5, 2020-12-03:
`  //初始化timer
    JL_TIMERx->CON = 0;
    JL_TIMERx->CON |= (0b10 << 2);						//选择晶振时钟源：24MHz
    JL_TIMERx->CON |= (0b0001 << 4);					//时钟源4分频
    JL_TIMERx->PRD = OSC_Hz / (4 * fre);				//设置周期
    JL_TIMERx->PWM = (JL_TIMERx->PRD * duty) / fre;	//0~10000对应0~100g
    JL_TIMERx->CNT = 0;									//清计数值
    JL_TIMERx->CON |= (0b01 << 0);						//计数模式

    if (hw_port == port) {
        gpio_set_die(hw_port, 1);
        gpio_set_pull_up(hw_port, 0);
        gpio_set_pull_down(hw_port, 0);
        gpio_set_direction(hw_port, 0);
        JL_TIMERx->CON |= BIT(8);						//PWM使能
    } else {

        gpio_output_channle(port, output_ch);
        return 1;
    }`
0.5版本SDK的../sdk/cpu/br25/mcpwm.c的487行，这是个明显的错误

maxbang5, 2020-12-03:
在br30上这样写pwm，还是不行
`{
    JL_TIMER2->CON = 0;
    JL_TIMER2->CON |= (0b10 << 2);						//选择晶振时钟源：24MHz
    JL_TIMER2->CON |= (0b0001 << 4);					//时钟源4分频
    JL_TIMER2->PRD = 1500;				                //设置周期
    JL_TIMER2->PWM = 750;//(JL_TIMER2->PRD * 5) / 4;	//0~10000对应0~100g
    JL_TIMER2->CNT = 0;									//清计数值
    JL_TIMER2->CON |= (0b01 << 0);						//计数模式

    gpio_set_die(IO_PORTB_06, 1);
    gpio_set_pull_up(IO_PORTB_06, 0);
    gpio_set_pull_down(IO_PORTB_06, 0);
    gpio_set_direction(IO_PORTB_06, 0);
    JL_TIMER2->CON |= BIT(8);						//PWM使能
}`

wjjjb123, 2020-12-03:
/**
 * @param JL_TIMERx : JL_TIMER0/1/2/3/4/5
 * @param pwm_io : JL_PORTA_01, JL_PORTB_02,,,等等，br30支持任意普通IO
 * @param fre : 频率，单位Hz
 * @param duty : 初始占空比，0~10000对应0~100%
 */
void timer_pwm_init(JL_TIMER_TypeDef * JL_TIMERx, u32 pwm_io, u32 fre, u32 duty)
{
	u32 io_clk = 0;
	gpio_set_die(pwm_io, 1);
	gpio_set_pull_up(pwm_io, 0);
	gpio_set_pull_down(pwm_io, 0);
	gpio_set_direction(pwm_io, 0);
	switch((u32)JL_TIMERx) {
		case (u32)JL_TIMER0 :
			JL_IOMAP->CON0 |= BIT(0);
			io_clk = 24000000;
			gpio_set_fun_output_port(pwm_io, FO_TMR0_PWM, 0, 1);
			break;
		case (u32)JL_TIMER1 :
			JL_IOMAP->CON0 |= BIT(1);
			io_clk = 12000000;
			gpio_set_fun_output_port(pwm_io, FO_TMR1_PWM, 0, 1);
			break;
		case (u32)JL_TIMER2 :
			JL_IOMAP->CON0 |= BIT(2);
			io_clk = 24000000;
			gpio_set_fun_output_port(pwm_io, FO_TMR2_PWM, 0, 1);
			break;
		case (u32)JL_TIMER3 :
			bit_clr_ie(IRQ_TIME3_IDX);
			JL_IOMAP->CON0 |= BIT(3);
			io_clk = 12000000;
			gpio_set_fun_output_port(pwm_io, FO_TMR3_PWM, 0, 1);
			break;
		case (u32)JL_TIMER4 :
			JL_IOMAP->CON0 |= BIT(4);
			io_clk = 24000000;
			gpio_set_fun_output_port(pwm_io, FO_TMR4_PWM, 0, 1);
			break;
		case (u32)JL_TIMER5 :
			JL_IOMAP->CON0 |= BIT(5);
			io_clk = 12000000;
			gpio_set_fun_output_port(pwm_io, FO_TMR5_PWM, 0, 1);
			break;
		default:
			return;
	}
	//初始化timer
	JL_TIMERx->CON = 0;
	JL_TIMERx->CON |= (0b01 << 2);						//时钟源选择IO_CLK
	JL_TIMERx->CON |= (0b0001 << 4);					//时钟源再4分频
	JL_TIMERx->CNT = 0;									//清计数值
	JL_TIMERx->PRD = io_clk / (4 * fre);				//设置周期
	JL_TIMERx->CON |= (0b01 << 0);						//计数模式
	JL_TIMERx->CON |= BIT(8);							//PWM使能
	//设置初始占空比
	JL_TIMERx->PWM = (JL_TIMERx->PRD * duty) / 10000;	//0~10000对应0~100%
}

void set_timer_pwm_duty(JL_TIMER_TypeDef * JL_TIMERx, u32 duty)
{
	JL_TIMERx->PWM = (JL_TIMERx->PRD * duty) / 10000;	//0~10000对应0~100%
}

wjjjb123, 2020-12-03:
目前br30 SDK 占用了0/1/2，用户可以使用timer3/4/5，后续可能会把timer0也提供出来，改成用户可用。
timer0在SDK里被用作了局部的us级延时。
timer1是SDK系统节拍
timer2在低功耗流程中被用作延时。

maxbang5, 2020-12-03:
那关于br30 的定时器中断，也是一样只能用timer3/4/5么，写法与br25的有区别么，用timer4写定时中断，测试无效，请问有可实现定时中断的例程么

maxbang5, 2020-12-03:
这边需要用到br30 的**定时中断发送数据**，需要一个例程

lawrencejiabin, 2020-12-04:
#include "includes.h"
#include "app_config.h"
#include "asm/cpu.h"
#include "asm/irq.h"
#include "asm/clock.h"
#include "system/timer.h"
#include "system/init.h"
#include "gpio.h"

/* #define LOG_TAG_CONST   TMR */
/* #define LOG_TAG         "[USER_TMR]" */
/* #define LOG_INFO_ENABLE */
/* #define LOG_DUMP_ENABLE */
/* #define LOG_ERROR_ENABLE */
/* #define LOG_DEBUG_ENABLE */
#include "debug.h"
/*
注意

timer 现在定义优先级为6 ，关总中断不关闭该优先级，
该中断里面使用函数 const 变量都必须定义在ram，否则会跑飞
 *
 */
#if 1
#define TCFG_USER_TIMER_ENABLE
#endif

#ifdef TCFG_USER_TIMER_ENABLE

struct timer_hdl {
    int index;
    int prd;
};

static struct timer_hdl hdl;

#define __this  (&hdl)

static const u32 timer_div[] = {
    /*0000*/    1,
    /*0001*/    4,
    /*0010*/    16,
    /*0011*/    64,
    /*0100*/    2,
    /*0101*/    8,
    /*0110*/    32,
    /*0111*/    128,
    /*1000*/    256,
    /*1001*/    4 * 256,
    /*1010*/    16 * 256,
    /*1011*/    64 * 256,
    /*1100*/    2 * 256,
    /*1101*/    8 * 256,
    /*1110*/    32 * 256,
    /*1111*/    128 * 256,
};

#define APP_TIMER_CLK           clk_get("lsb")//clk_get("timer")
#define MAX_TIME_CNT            0x7fff
#define MIN_TIME_CNT            0x100

#define TIMER_CON               JL_TIMER4->CON
#define TIMER_CNT               JL_TIMER4->CNT
#define TIMER_PRD               JL_TIMER4->PRD
#define TIMER_VETOR             IRQ_TIME4_IDX

#define TIMER_UNIT_MS           1 //1ms起一次中断
#define MAX_TIMER_PERIOD_MS     (1000/TIMER_UNIT_MS)

/*-----------------------------------------------------------*/
static void (*timer_led_scan)(void *param);

void app_timer_led_scan(void (*led_scan)(void *))
{
    timer_led_scan = led_scan;
}

/////下面函数调用的使用函数都必须放在ram
AT_VOLATILE_RAM_CODE
/* static void user_timer_isr() */
___interrupt
static void timer4_isr()
{
    TIMER_CON |= BIT(14);
    /* putchar('T'); */
    if (timer_led_scan) {
        timer_led_scan(NULL);
    }
    static u8 flag = 0;
    if(flag){
        flag = 0;
        /* gpio_set_output_value(IO_PORTA_03, 0); */
        JL_PORTA->DIR &= ~BIT(3);
        JL_PORTA->OUT &= ~BIT(3);
    }else{
        flag = 1;
        /* gpio_set_output_value(IO_PORTA_03, 1); */
        JL_PORTA->DIR &= ~BIT(3);
        JL_PORTA->OUT |= BIT(3);
    }
}

int user_timer_init()
{
    u32 prd_cnt;
    u8 index;

    printf("------------%s :%d", __func__, __LINE__);
    for (index = 0; index < (sizeof(timer_div) / sizeof(timer_div[0])); index++) {
        prd_cnt = TIMER_UNIT_MS * (APP_TIMER_CLK / 10000) / timer_div[index];
        if (prd_cnt > MIN_TIME_CNT && prd_cnt < MAX_TIME_CNT) {
            break;
        }
    }
    /* index = 0b1101; */
    __this->index   = index;
    __this->prd     = prd_cnt;

    TIMER_CNT = 0;
    TIMER_PRD = prd_cnt; //1ms
    request_irq(TIMER_VETOR, 1, timer4_isr, 0);
    TIMER_CON = BIT(14) | (index << 4) | BIT(0);//| BIT(3)

    r_printf("LSB CLK:%d", clk_get("lsb"));
    r_printf("CON:0x%x  PRD:0x%x  CLK:%d",TIMER_CON, TIMER_PRD, clk_get("timer"));

    gpio_set_pull_up(IO_PORTA_03, 0);
    gpio_set_pull_down(IO_PORTA_03, 0);
    gpio_set_die(IO_PORTA_03, 1);
    gpio_set_direction(IO_PORTA_03, 0);

    return 0;
}
/* __initcall(user_timer_init); */

void printf_timer_log(void)
{
    printf("irq_read(TIMER_VETOR):%d",irq_read(TIMER_VETOR));
    printf("irq_check_ipnd(TIMER_VETOR):%d",irq_check_ipnd(TIMER_VETOR));
    printf("JL_TIMER4->CON:0x%x",TIMER_CON);
    printf("JL_TIMER4->PRD:0x%x",TIMER_PRD);
    printf("JL_TIMER4->CNT:0x%x",TIMER_CNT);
    /* bit_set_ie(TIMER_VETOR); */
    printf("\n\n");
}

#endif

forrestan, 2020-12-14:
可否用IO模拟PWM?一个定时器，多个IO同时输出PWM?

closed 2021-05-08 by Tsai1005