Jieli-Tech/fw-AC63_BT_SDK #30: AC637N实现硬件触摸

maxbang5, 2020-12-08:
功能：在AC637N上用四个IO口实现四个硬件触摸，但SDK中没有对应的硬件触摸驱动，“lp_touch_key.c”和“lp_touch_key.h”
需求：麻烦提供实现硬件触摸所需要的文件，谢谢

rixin-Tech, 2020-12-11:
AC637N支持两路触摸。下个版本更新驱动。

maxbang5, 2020-12-11:
这边不需要触摸唤醒，不需要考虑功耗，我看软件中触摸是分为touch和ctmu_touch，普通触摸可以实现四个触摸么

rixin-Tech, 2020-12-11:
可以，有8 普通的触摸按键个通道
[image]

maxbang5, 2020-12-14:
这些触摸对应IO口是固定的么，可以复用么，麻烦提供下demo，文件较大可以发邮件 xuebangma@163.com

rixin-Tech, 2020-12-14:
[8a448fe.diff.zip](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/5687738/8a448fe.diff.zip)

maxbang5, 2020-12-14:
这个.diff的补丁文件没用过，如何使用？

maxbang5, 2020-12-14:
这个是在0.5版本sdk上修改的么，0.5sdk里没有“ctmu_touch_key.c”这个文件，有没有完整的sdk

rixin-Tech, 2020-12-15:
完整的sdk后续发布，0.6版本支持一路低功耗触摸，8路普通触摸

maxbang5, 2020-12-15:
0.6版本的SDK是什么时候发布，能不能提供下时间，太晚的话可不可以先提供一个可以使用的sdk版本

maxbang5, 2020-12-17:
硬件：AC6373B
软件：AC630N_bt_data_transfer_sdk_release_v0.6.0
问题：开启 TCFG_CTMU_TOUCH_KEY_ENABLE 使能后，测试触摸，发现触摸短按没问题，长按容易检测不到抬起；而 TCFG_LP_TOUCH_KEY_ENABLE  使能后的低功耗触摸口没这个问题，希望解决普通触摸长按检测不到抬起的问题

rixin-Tech, 2020-12-17:
你的触摸按键是怎么设计的，需要带个外壳才行。

maxbang5, 2020-12-17:
普通触摸IO口接飞线接铜箔，触摸铜箔后，短按没问题，长按就很容易检测不到长按，AC6373B上两个普通触摸的IO口现象是一样的；
同样的方式在低功耗触摸的IO口上操作，短按长按都是没问题的；
目前我这边是PB1的低功耗触摸没问题，但是两个普通触摸PA3、PB6出现的问题都一致，带外壳触摸也有不正常的现象，所以感觉不是结构上的问题，目前没画板，是在开发板上飞线的

maxbang5, 2020-12-21:
专门画了个触摸的板子，走线按照要求走的，现象和原本测试的一致，普通触摸还是容易检测不到抬起，麻烦帮忙测试下0.6版本sdk，我软件用的是原始sdk，只打开了低功耗触摸和普通触摸的使能

closed 2021-05-08 by Tsai1005