Jieli-Tech/fw-AC63_BT_SDK #31: 输出数据会被无规律打断

forrestan, 2020-12-09:
#define WS_OUT	IO_PORTA_08//IO_PORTB_08
void Ws_SendByte(u8 data)
{
	u8 i=8,j;
	for(i=0; i<8; i++)
	{
		if(data&0x80)  //send 1
		{
			gpio_write(WS_OUT, 1);
			delay(0);
			gpio_write(WS_OUT, 0);
			data<<=1; 		 //左移一位
		}
		else // send 0
		{
			gpio_write(WS_OUT, 1);
			data<<=1; 	//左移一位
			gpio_write(WS_OUT, 0);
			delay(0);
		}
	}
}
void Ws_play()
{
  int i;
	
	for(i=0;i<6;i++)
	{
		Ws_SendByte(RainBow[i]);
		Ws_SendByte(RainBow[i]);
		Ws_SendByte(RainBow[i]);
	}

}
正确波形

[image]

错误波形：

[image]
[image]
[image]

rixin-Tech, 2020-12-09:
关闭中断

forrestan, 2020-12-11:
关闭哪个中断？

rixin-Tech, 2020-12-11:
local_irq_disable();
//TODO：

local_irq_enable();

forrestan, 2020-12-11:
看到了，谢谢。

rixin-Tech, 2020-12-11:
对时间敏感，切消耗时间比较久的波形可以使用spi do 来dma输出

closed 2020-12-17 by rixin-Tech