Jieli-Tech/fw-AC63_BT_SDK #33: 关于flash分布

stronges, 2020-12-16:
可以给个内存布局图么？ 
[image]

[image]
这是有没有开ble ota对比，只能看到vm和flash减少， 入口地址是0x1E00120， 那么从上图来看，FLASH_BIN_SIZE的起始地址应该是0,  根据你们所说，boot的起始地址为1e000100， 那么boot有多大了？

rixin-Tech, 2020-12-16:
1e000100 是逻辑地址，你对比一下芯片的memory 介绍

stronges, 2020-12-16:
> 1e000100 是逻辑地址，你对比一下芯片的memory 介绍

ac6368A, 哪里有memery介绍，datasheet上没有

rixin-Tech, 2020-12-17:
[image]

xjtuecho, 2021-05-01:
ac6368a是否有详细一些的用户手册？datasheet太简略信息太少了。

Tsai1005, 2021-05-08:
@xjtuecho 用户手册不支持线上发布，请联系代理商和FAE

closed 2021-05-08 by Tsai1005

xjtuecho, 2021-05-17:
> @xjtuecho 用户手册不支持线上发布，请联系代理商和FAE

代理商说没有datasheet，就是这个仓库里面的那个9页的pdf

rixin-Tech, 2021-12-06:
[image]