Jieli-Tech/fw-AC63_BT_SDK #37: [AC6311]   如何通过定时器实现GPIO引脚（PB9）的脉冲宽度检测， 请给出demo 。

tylerYu123, 2021-01-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

tylerYu123, 2021-01-07:
基于AC6311芯片， 通过定时器实现GPIO引脚（PB9）的脉冲宽度检测功能 ，可以设置检测精度 。  需要提供Demo程序  。

rixin-Tech, 2021-01-07:
#define GPCNT_EN(x) SFR(JL_GPCNT->CON, 0, 1, x)
#define GPCNT_CSS(x) SFR(JL_GPCNT->CON, 1, 3, x)

enum {
    GPCNT_CSS_LSB = 0,
    GPCNT_CSS_OSC,
    GPCNT_CSS_CAP_MUX,
    GPCNT_CSS_CLK_MUX,
    GPCNT_CSS_GPCFD,
    GPCNT_CSS_RING,
    GPCNT_CSS_PLL480M,
    GPCNT_CSS_IRFLT,
};

#define GPCNT_CLR_PEND(x) SFR(JL_GPCNT->CON, 6, 1, x)
#define GPCNT_GTS(x) SFR(JL_GPCNT->CON, 8, 4, x)
#define GPCNT_GSS(x) SFR(JL_GPCNT->CON, 12, 3, x)

enum {
    GPCNT_GSS_LSB = 0,
    GPCNT_GSS_OSC,
    GPCNT_GSS_CAP_MUX,
    GPCNT_GSS_CLK_MUX,
    GPCNT_GSS_GPCFD,
    GPCNT_GSS_RING,
    GPCNT_GSS_PLL480M,
    GPCNT_GSS_IRFLT,
};
#define MHz_UNIT 1000000L
#define KHz_UNIT 1000L
#define CY 0x2
SET_INTERRUPT
void gpcnt_isr()
{
    JL_GPCNT->CON |= BIT(6);
    volatile int xosc = 192 * MHz_UNIT / JL_GPCNT->NUM * (32 * (2 << (CY - 1)));

    GPCNT_EN(0);
    GPCNT_EN(1);
}
void gpcnt_init()
{
	JL_PORTB->DIR |= BIT(9);
    JL_PORTB->DIE |= BIT(9);
    JL_PORTB->PU &= ~BIT(9);
    JL_PORTB->PD &= ~BIT(9);
    JL_IOMAP->CON2 &= ~(0x3f << 16); 
	JL_IOMAP->CON2 |= (IO_PORTB_09 << 16); //PB9--->input ch2

    GPCNT_CLR_PEND(1);
    GPCNT_GSS(0b010);
    GPCNT_CSS(0b110);
    GPCNT_GTS(CY);

    GPCNT_CLR_PEND(1);
    GPCNT_EN(1);

    HWI_Install(GPCNT_INIT, (int)gpcnt_isr, 1);
}

rixin-Tech, 2021-01-07:
[image]

tylerYu123, 2021-01-08:
[image]

tylerYu123, 2021-01-08:
请问主时钟与次时钟是如何同步的 ？  我的理解是芯片捕捉到次时钟上升沿后开始计时， 是这样吗 ？

tylerYu123, 2021-01-08:
请提供SFR（）， HWI_Install（） 头文件 。

rixin-Tech, 2021-01-11:
1、#define SFR(sfr, start, len, dat) \
	(sfr = (sfr & ~((~(0xffffffff << (len))) << (start))) | \
	 (((dat) & (~(0xffffffff << (len)))) << (start)))

2、HWI_Install 和 request_irq 功能相同，注册中断函数。

rixin-Tech, 2021-01-11:
> 请问主时钟与次时钟是如何同步的 ？ 我的理解是芯片捕捉到次时钟上升沿后开始计时， 是这样吗 ？

是的。计时结束，产生中断

closed 2021-01-15 by tylerYu123