Jieli-Tech/fw-AC63_BT_SDK #39: 关于系统重启

tylerYu123, 2021-01-11:
在后台任务或定时器中断中调用哪个API能重启系统 ？ 同时提供重启后get_power_reset_source_info（）的返回值 。 谢谢！

rixin-Tech, 2021-01-11:
调用chip_reset(); 
get_power_reset_source_info这个函数区分不了主动复位。
在复位前存状态到vm。开机后读取。

tylerYu123, 2021-01-11:
看门狗复位是可以检测到吧？

rixin-Tech, 2021-01-11:
程序跑错也可能出现看门狗复位，你需要读取这个状态来实现什么功能？

tylerYu123, 2021-01-11:
提示机子异常了。

rixin-Tech, 2021-01-11:
[image]

rixin-Tech, 2021-01-11:
VDDIO POR 是首次上电，即芯片从没电到有电，从头炮
VDDIO LVD 是电池电量不够或者VDDIO被扯，导致VDDIO 的电压低于设定的阈值，复位了
WDT 是看门狗超时，复位了
VCM 是芯片引脚，默认的短按就会复位，就是通常板子上的reset键就是用这个脚做的。
PPINR 是设置的长按复位脚，长按引起的复位
SYS RESET 是内核系统的复位，跟内核系统有关，默认关闭的，如果打开的话，比如内核电源DVDD有上下电，就会引起内核系统复位。
SOFT RESET 是P33系统管理的复位，一般是程序里执行了复位函数，如果代码异常了，异常中断里会执行复位函数来复位芯片，比如一些非法操作：除0异常，非法地址访问等等。

tylerYu123, 2021-01-12:
测试看门狗重启（后台任务跑while（1） ， 不喂狗 ）， 打印出的原因是0xC0 。 请问是什么原因？

rixin-Tech, 2021-01-12:
提供完整log。0xc0是软复位(SOFT RESET)。

tylerYu123, 2021-01-12:
void setup_arch()
{
    wdt_init(WDT_4S);

    // 解锁芯片IO     乐心添加 按键唤醒时有log输出
    p33_tx_1byte(P3_ANA_LAT, 0x00);

    clk_voltage_init(TCFG_CLOCK_MODE, 1160);

    clk_early_init(TCFG_CLOCK_SYS_SRC, TCFG_CLOCK_OSC_HZ, TCFG_CLOCK_SYS_HZ);

#if (defined CONFIG_DEBUG_ENABLE) || (defined CONFIG_DEBUG_LITE_ENABLE)

    JL_UART0->CON0 |= BIT(0);
    JL_UART1->CON0 |= BIT(0);

    debug_uart_init(NULL);

    JL_UART0->CON0 &= ~BIT(0);
    JL_UART1->CON0 &= ~BIT(0);

    /* void vmm_debug(); */
    /* vmm_debug(); */

    /*interrupt_init();*/

#ifdef CONFIG_DEBUG_ENABLE
    log_early_init(1024);
#endif

#endif
    log_i("\n~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~");
    log_i("         setup_arch %s %s", __DATE__, __TIME__);
    log_i("~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~\n");

    clock_dump();

    /* log_info("resour est: %d", get_boot_flag()); */
    //set_boot_flag(99);
    /* log_info("resour est: %d", get_boot_flag()); */

    reset_source_dump();

    power_reset_source_dump();

    request_irq(1, 2, exception_irq_handler, 0);

    debug_init();

    sys_timer_init();

    /* sys_timer_add(NULL, timer, 10 * 1000); */

    tick_timer_init();

    __crc16_mutex_init();

    // 乐心 app任务上电&唤醒 初始化
    LSApp_InitSystem();

    while ( 1 )   // 在setup_arch（）中加while测试看门狗复位 
    {
         
    }
   
}

tylerYu123, 2021-01-12:
[00:00:00.100][Info]: [MISC]--Reset Source : 0x1
[00:00:00.100][Info]: [MISC]P33 RESET
[00:00:00.100][Info]: [LSMAIN]
 ************** Start to Init Ls System  ***********************

[00:00:00.100][Info]: [HW_DBG]reset reason = 0x81
[00:00:00.100][Info]: [HW_DBG]reset type = 0x0
[00:00:00.100][Info]: [Timer]LsSystimer_Init: clk timer = 24000000
[00:00:00.100][Info]: [LSMAIN]
 ************** End of Init Ls System  ***********************

[00:00:00.100][Info]: [MISC]--Reset Source : 0x1
[00:00:00.100][Info]: [MISC]P33 RESET
[00:00:00.100][Info]: [LSMAIN]
 ************** Start to Init Ls System  ***********************

[00:00:00.100][Info]: [HW_DBG]reset reason = 0xc0
[00:00:00.100][Info]: [HW_DBG]reset type = 0x1
[00:00:00.100][Info]: [Timer]LsSystimer_Init: clk timer = 24000000
[00:00:00.100][Info]: [LSMAIN]
 ************** End of Init Ls System  ***********************

[00:00:00.100][Info]: [MISC]--Reset Source : 0x1
[00:00:00.100][Info]: [MISC]P33 RESET
[00:00:00.100][Info]: [LSMAIN]
 ************** Start to Init Ls System  ***********************

[00:00:00.100][Info]: [HW_DBG]reset reason = 0xc0
[00:00:00.100][Info]: [HW_DBG]reset type = 0x1
[00:00:00.100][Info]: [Timer]LsSystimer_Init: clk timer = 24000000
[00:00:00.100][Info]: [LSMAIN]
 ************** End of Init Ls System  ***********************

tylerYu123, 2021-01-12:
首次上电复位原因是0x81 ; 看门狗复位是0xC0 。

tylerYu123, 2021-01-16:
请更新一版库文件， 修复看门狗复位原因代码为设计值 .

closed 2021-05-08 by Tsai1005