Jieli-Tech/fw-AC63_BT_SDK #43: 关于AC635N的sniff

maxbang5, 2021-01-28:
芯片仅开了ble，在sniff时唤醒后处理完，多长时间会重新进入sniff，这个时间可调整么，修改哪个参数调整

hhy-ccj, 2021-02-03:
系统进入idel，蓝牙不需要工作时重新进入powerdown

maxbang5, 2021-02-03:
在bt模式下，中断唤醒后进入重新powerdown需要50ms，我唤醒后就跑了几行代码，后面这么长时间再进入powerdown，是不可更改的么，只有在idle的时候才能短时间进入么，但是这边是需要ble的

hhy-ccj, 2021-02-03:
唤醒后除了跑用户自定义代码，可能会有按键扫描等外设，定时器这些，让系统处于忙的状态

maxbang5, 2021-02-03:
这边示波器测试，在idle下，外部中断唤醒，也是50ms左右重新进入powerdown，仅跑了外部中断，没有使能按键和定时器，这个是为什么呢，不能跑完代码立马进入powerdown么，

hhy-ccj, 2021-02-03:
是哪个版本sdk，哪个系列

maxbang5, 2021-02-03:
用的SDK是“AC630N_bt_data_transfer_sdk_release_v0.6.0的spp_and_le的br23”，唤醒的写法是这种，添加了外部中断，唤醒后进入外部中断处理，现在是处理完外部中断代码，明显不能快速进入powerdown；
硬件IC是AC6351D；
/********定义唤醒口*********/
struct port_wakeup port0 = {
	.pullup_down_enable = DISABLE,                            //配置I/O 内部上下拉是否使能
	.edge               = RISING_EDGE,                      //唤醒方式选择,可选：上升沿\下降沿
	.attribute          = BLUETOOTH_RESUME,                  //保留参数
	.iomap              = IO_PORTB_07,                         //唤醒口选择
    .filter_enable      = DISABLE,
};

const struct sub_wakeup sub_wkup = {
	.attribute  = BLUETOOTH_RESUME,
};

const struct charge_wakeup charge_wkup = {
	.attribute  = BLUETOOTH_RESUME,
};

const struct wakeup_param wk_param = {
	.port[1] = &port0,
	.sub = &sub_wkup,
	.charge = &charge_wkup,
};
/*********初始化外部中断*************/
port_wkup_interrupt_init(IO_PORTB_07, rising_edge_trigger, io_isr_cbfun1);
中断函数中仅简单计数，没有循环；

hhy-ccj, 2021-02-03:
是ble单模吗，有使能edr吗

maxbang5, 2021-02-03:
能使能，但是没有使能，而且不在bt模式下，在idle模式下也是50ms左右

maxbang5, 2021-02-04:
你可以用原始sdk测试下，应该都是一样的，唤醒后要50ms左右再进入powerdown，这个时间是比较长的，现在这个可以修改么，希望改短

Leewoniu, 2021-02-19:
可以使用 用的SDK是“AC630N_bt_data_transfer_sdk_release_v0.7.0的版本，不存在这个问题

Leewoniu, 2021-02-19:
> 你可以用原始sdk测试下，应该都是一样的，唤醒后要50ms左右再进入powerdown，这个时间是比较长的，现在这个可以修改么，希望改短

可以使用 用的SDK是“AC630N_bt_data_transfer_sdk_release_v0.7.0的版本，不存在这个问题

hhy-ccj, 2021-02-19:
070 SDK已解决

closed 2021-02-19 by hhy-ccj