Jieli-Tech/fw-AC63_BT_SDK #49: 696x 1.2sdk ， 想用硬件定时器做较为准确的us_delay ，发现定时器时间很不均匀，如何解决？

REYNJHR, 2021-02-25:
下图是io在中断里取反：[image]

如下是定时器配置：
#if (1)
#define TCFG_USER_TIMER_ENABLE
#endif

#ifdef TCFG_USER_TIMER_ENABLE

struct timer_hdl {
    int index;
    int prd;
};

static struct timer_hdl hdl;

#define __this  (&hdl)

static const u32 timer_div[] = {
    /*0000*/    1,
    /*0001*/    4,
    /*0010*/    16,
    /*0011*/    64,
    /*0100*/    2,
    /*0101*/    8,
    /*0110*/    32,
    /*0111*/    128,
    /*1000*/    256,
    /*1001*/    4 * 256,
    /*1010*/    16 * 256,
    /*1011*/    64 * 256,
    /*1100*/    2 * 256,
    /*1101*/    8 * 256,
    /*1110*/    32 * 256,
    /*1111*/    128 * 256,
};

#define APP_TIMER_CLK           clk_get("timer")
#define MAX_TIME_CNT            0x7fff
#define MIN_TIME_CNT            0x100

#define TIMER_CON               JL_TIMER3->CON
#define TIMER_CNT               JL_TIMER3->CNT
#define TIMER_PRD               JL_TIMER3->PRD
#define TIMER_VETOR             IRQ_TIME3_IDX

#define TIMER_UNIT_MS           1 //1ms起一次中断
#define MAX_TIMER_PERIOD_MS     (1000/TIMER_UNIT_MS)

/*-----------------------------------------------------------*/

/////下面函数调用的使用函数都必须放在ram
volatile unsigned long cut =0;
volatile unsigned char cut_flag=0;

___interrupt
AT_VOLATILE_RAM_CODE
static void timer3_isr()
{
//    static u32 cnt1 = 0;

    TIMER_CON |= BIT(14);
//     if(cut_flag)
//    {
//        cut++;
//    }
  JL_PORTB->OUT ^= BIT(7);
//

}
void timer2_delay(unsigned int time)
{
    cut_flag=1;
    cut=0;
    while(1)
    {
        if(cut>time)
            break;
    }
    cut_flag =0;
}
int timer3_init()
{
    u32 prd_cnt;
    u8 index;

    printf("------------%s :%d", __func__, __LINE__);

    for (index = 0; index < (sizeof(timer_div) / sizeof(timer_div[0])); index++) {
        prd_cnt = TIMER_UNIT_MS * (APP_TIMER_CLK / 6000) / timer_div[index];
        if (prd_cnt > MIN_TIME_CNT && prd_cnt < MAX_TIME_CNT) {
            break;
        }
    }
    __this->index   = index;
    __this->prd     = prd_cnt;

    TIMER_CNT = 0;
    TIMER_PRD = prd_cnt; //1ms
    request_irq(TIMER_VETOR, 1, timer3_isr, 0);
    TIMER_CON = (index << 4) | BIT(0) | BIT(3);

    printf("PRD : 0x%x / %d", TIMER_PRD, clk_get("timer"));

    return 0;
}
__initcall(timer3_init);

rixin-Tech, 2021-02-25:
如果是需要跟外部通讯，可以通过spi do来输出准确的波形。
做udelay来延时， 关闭中断，使用查询pnd方式。

REYNJHR, 2021-02-25:
> 抄　送：姜浩睿 <jianghr@meshcore.cn>; Author <author@noreply.github.com>
> 主　题：Re: [Jieli-Tech/fw-AC63_BT_SDK] 696x 1.2sdk ， 想用硬件定时器做较为准确的us_delay ，发现定时器时间很不均匀，如何解决？ (#49)

能否给个例程，或者sdk里哪有类似的例子？

------------------------------------------------------------------

rixin-Tech, 2021-02-25:
AT_VOLATILE_RAM_CODE
void udelay(u32 us)
{
	local_irq_disable();
	JL_TIMER2->PRD =  24000000/1000000*us;
	JL_TIMER2->CNT = 0;                                                      
       JL_TIMER2->CON = CLR_PND | (BT_OSC_CLK << 2) | BIT(0);     // btosc 24M  
	while(1){
		if(JL_TIMER2->CON & BIT(15)){
			break;
		}
	}
	local_irq_enable();
}

closed 2021-05-08 by Tsai1005