Jieli-Tech/fw-AC63_BT_SDK #51: 更新0.7.1SDK后蓝牙地址问题

Williamzhangtw, 2021-02-26:
从0.5SDK更新到0.7.1SDK后. 编译工程得到update.ufw 升级文件并放到云服务器上。 设备通过OTA升级update.ufw.  升级后出现了问题。 

BLE蓝牙MAC地址改变了.(例如原来最后两位是35，变为了23)

yzb201611, 2021-03-01:
确认isd_config.ini文件的BTIF_OPT=0；等于0会擦掉蓝牙地址。或者0.7.1和0.5的sdk存储结构略有区别也会出现btif区域位置变化导致原来的数据丢失。确认下0.7.1是否升级0.7.1就不会有这个现象。

Williamzhangtw, 2021-03-04:
> 确认isd_config.ini文件的BTIF_OPT=0；等于0会擦掉蓝牙地址。或者0.7.1和0.5的sdk存储结构略有区别也会出现btif区域位置变化导致原来的数据丢失。确认下0.7.1是否升级0.7.1就不会有这个现象。
> 
我這邊驗證了0.7版本的後面升級的沒有出現這個問題。
我擔心的是
0.7版本的升級到以後比如0.8版本的還會不會出現同樣的問題？比如說我做好了0.7的發送到客戶手上如果升級0.8出現這個問題客戶升級後產品就變成磚頭了,要寄回才能修了。

我這邊需要怎麼設計才能避免這個問題？

mikewen, 2021-03-12:
如果每个芯片有 unique chip id 的话, 就可以解决:
App OTA 之前, save chip id->mac  pair, after OTA 之后, scan->new mac->read chip id,  
然后 either 更新 App, 或 chip 重写 old mac ( bt_mac_addr_set )

closed 2021-05-08 by Tsai1005