Jieli-Tech/fw-AC63_BT_SDK #53: 使能音量同步后，通过按键将音量减到没有声音后，继续按音量减，此时再按音量加后发现音量就再也加不上去了。只有切换模式或者重启音量才能恢复正常。什么原因？如何解决？

REYNJHR, 2021-02-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

REYNJHR, 2021-02-27:
696x  1.2sdk

Tsai1005, 2021-05-08:
音频系列SDK 请联系代理商或FAE 线下支持

closed 2021-05-08 by Tsai1005