Jieli-Tech/fw-AC63_BT_SDK #55: nRF Mesh unable to Connect node after provisioned

mikewen, 2021-03-05:
**Describe the bug**
After using 'nRF Mesh' Android App to 'Add Node', unable to connect after disconnect the node.

**To Reproduce**
Steps to reproduce the behavior:
1. Use 'nRF Mesh' to 'Add Node', bind key.
2. Disconnect or exit the 'nRF Mesh' App
3. Click 'connect'
4. Scanner keep looking but unable to find the provisioned nodes

**Expected behavior**
'nRF Mesh' App should be able to find provisioned nodes

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: LG G7
 - OS: Android
 - Browser [e.g. chrome, safari]
 - SDK Version, tried both 0.7.0 and 0.8.0

hhy-ccj, 2021-03-08:
是哪个系列的芯片，nRF Mesh是哪个版本

NoCodeNoLife-404, 2021-03-08:
[image]
将lib_btctrler_config.c文件中的ble_disable_wait_enable改为0

mikewen, 2021-03-08:
AC6369F,  'nRF Mesh' 是 3.1.0 
试了 ble_disable_wait_enable = 0，还是不行

NoCodeNoLife-404, 2021-03-08:
是否有尝试清空flash和rebuild代码？

mikewen, 2021-03-08:
rebuild代码 试过了, 不行.  怎样清空flash?

NoCodeNoLife-404, 2021-03-08:
https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/releases/download/v2.4.1/nRF.Mesh.-.V.-.2.4.1.apk
请使用这个版本的nRF Mesh测试下

mikewen, 2021-03-09:
这个版本的nRF Mesh测试可以了!  多谢!

hhy-ccj, 2021-03-09:
> 这个版本的nRF Mesh测试可以了! 多谢!

对于新版本软件不适配的问题，我们后续会跟进。

hhy-ccj, 2021-03-09:
nRF Mesh 新版本 3.1.0不适配，稍后解决

closed 2021-03-09 by hhy-ccj