Jieli-Tech/fw-AC63_BT_SDK #62: AC6969C为MCU的mesh设备通讯丢码问题

xxyang20, 2021-03-18:
1、AC630N_release_v0.8.0版本SDK，AC6969C为MCU的单个SIG Generic OnOff Client设备使用按键，通过mesh控制一个SIG Generic OnOff Server设备会出现丢码现象(Client设备按按键，Server设备未响应按键消息)。release_v0.3.0版本SDK未出现此现象。请问为什么会出现丢码现象？如何解决？
2、AC630N_release_v0.3.0版本SDK,AC6969C为MCU的单个SIG Generic OnOff Client设备通过mesh控制15个SIG Generic OnOff 
Server设备，连续用按键控制所有Server设备led灯亮灭30次，会陆续出现有Server设备不再响应Client设备按键消息，Server设备重新开机可以继续响应Client设备按键消息。为什么会出现Server设备不再响应mesh消息？

hhy-ccj, 2021-03-18:
> 1、AC630N_release_v0.8.0版本SDK，AC6969C为MCU的单个SIG Generic OnOff Client设备使用按键，通过mesh控制一个SIG Generic OnOff Server设备会出现丢码现象(Client设备按按键，Server设备未响应按键消息)。release_v0.3.0版本SDK未出现此现象。请问为什么会出现丢码现象？如何解决？ [...]

问题1：
080 SDK在scan_core.c文件修改如下：
#define MESH_SCAN_INTERVAL_MS   7
#define MESH_SCAN_WINDOW_MS   4

问题2：
在修复问题1基础上，尝试使用 080 SDK去测试

hhy-ccj, 2021-03-18:
> 1、AC630N_release_v0.8.0版本SDK，AC6969C为MCU的单个SIG Generic OnOff Client设备使用按键，通过mesh控制一个SIG Generic OnOff Server设备会出现丢码现象(Client设备按按键，Server设备未响应按键消息)。release_v0.3.0版本SDK未出现此现象。请问为什么会出现丢码现象？如何解决？ [...]

### 问题1：
**080 SDK** 在 **scan_core.c** 文件修改如下：
`
#define MESH_SCAN_INTERVAL_MS 7
#define MESH_SCAN_WINDOW_MS 4
`

### 问题2：
在修复 _问题1_ 基础上，尝试使用 **080 SDK** 去测试
如果还存在类似问题，如果是使用 **generic_onoff_client.c** 或者 **generic_onoff_server.c**，尝试修改 
**mesh_config_common.c**
`
#define MESH_ADV_BUFFER_COUNT           10 // 或者更大
`

如果以上问题还存在，再尝试修改mesh_config.h
`
#define CONFIG_BUF_REPLACE_EN	       1
`

hhy-ccj, 2021-03-26:
长时间没有得到回复，暂时关闭，如果问题仍存，请重新打开。

xxyang20, 2021-03-26:
1、AC630N_release_v0.8.0版本SDK丢码问题仍然没解决，修改MESH_SCAN_INTERVAL_MS、MESH_SCAN_WINDOW_MS可以降低丢码率，但是丢码仍然存在。Generic OnOff Client发单个消息是发两遍的，即便如此仍存在丢码率高的情况，这个问题怎么解决？
2、如果单个mesh设备要每秒获取一次每个设备状态，在设备总数50个甚至更多情况下，数据可以顺利传输且保证丢码率低吗？

hhy-ccj, 2021-03-26:
> 1、AC630N_release_v0.8.0版本SDK丢码问题仍然没解决，修改MESH_SCAN_INTERVAL_MS、MESH_SCAN_WINDOW_MS可以降低丢码率，但是丢码仍然存在。Generic OnOff Client发单个消息是发两遍的，即便如此仍存在丢码率高的情况，这个问题怎么解决？
> 2、如果单个mesh设备要每秒获取一次每个设备状态，在设备总数50个甚至更多情况下，数据可以顺利传输且保证丢码率低吗？

0.3.0版本SDK没有出现这样的问题？除了SDK不一样其它环境都一样吗？同样个数的设备，同样的处理方法？

xxyang20, 2021-03-26:
030版本sdk，单个Generic OnOff Client设备控制50个server设备也会出现丢码问题。

hhy-ccj, 2021-03-26:
> 030版本sdk，单个Generic OnOff Client设备控制50个server设备也会出现丢码问题。

- ### 尝试修改 client 发送消息类型为 unack
[image]

xxyang20, 2021-03-26:
好的，测试后没有发现丢码。谢谢！！！

closed 2021-03-26 by xxyang20