Jieli-Tech/fw-AC63_BT_SDK #64: 636N 的SDK 怎么开启BLE 5.0

ice108, 2021-03-24:


hhy-ccj, 2021-03-24:
## Core 5.0 特性
----------------

ble 的特性都由 _**config_btctler_le_features**_ 控制，位于相应 case 的 _**lib_btctrler_config.c**_，对于 core 5.0，新加了特性如下

```c
#define LE_CORE_V50_FEATURES \
( \
    LE_2M_PHY | \
    LE_CODED_PHY | \
    LE_EXTENDED_ADVERTISING | \
    LE_PERIODIC_ADVERTISING | \
    CHANNEL_SELECTION_ALGORITHM_2 | \
    0 \
)
```

## PHY 特性使用
----------------

- 如果要使用相应特性，需要在 _**config_btctrler_le_features**_ 或上相应位，例如想用 _**LE_EXTENDED_ADVERTISING**_，需要如下操作
```c
const uint64_t config_btctler_le_features = LE_ENCRYPTION  | LE_EXTENDED_ADVERTISING;
```

- 对于PHY的改变，如果作为从机，要连接的对象是支持 5.0 特性的手机，只要如下操作：
```c
// 手机自动更新连接PHY为2M
const uint64_t config_btctler_le_features = LE_ENCRYPTION |  LE_2M_PHY;
```
```c
// 手机自动更新连接PHY为coded
const uint64_t config_btctler_le_features = LE_ENCRYPTION | LE_CODED_PHY;
```

- 如果两个设备都是自家的，其中一个是我司的设备，除了按照上述方法修改 _**config_btctrler_le_features**_ 外，建议在连接完成后的事件回调函数主动发起 phy update，调用 _**set_connection_data_phy**_，操作如下：
```c
static void cbk_packet_handler(uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size)
{
    // ... ...
    switch (packet_type) {
    case HCI_EVENT_PACKET:
        switch (hci_event_packet_get_type(packet)) {
        case HCI_EVENT_LE_META:
            switch (hci_event_le_meta_get_subevent_code(packet)) {
            case HCI_SUBEVENT_LE_CONNECTION_COMPLETE: // 连接成功回调
                set_connection_data_phy(CONN_SET_2M_PHY, CONN_SET_2M_PHY); // 发起 phy update
                break;

            case HCI_SUBEVENT_LE_PHY_UPDATE_COMPLETE: // phy update 成功回调
                log_info("APP HCI_SUBEVENT_LE_PHY_UPDATE %s\n", hci_event_le_meta_get_phy_update_complete_status(packet) ? "Fail" : "Succ");
                log_info("Tx PHY: %s\n", phy_result[hci_event_le_meta_get_phy_update_complete_tx_phy(packet)]);
                log_info("Rx PHY: %s\n", phy_result[hci_event_le_meta_get_phy_update_complete_rx_phy(packet)]);
                break;

}
```

   更新成功后会在 _**HCI_SUBEVENT_LE_PHY_UPDATE_COMPLETE**_ 打印更新的结果。

ice108, 2021-03-25:
做client的时候调用set_connection_data_phy(CONN_SET_2M_PHY, CONN_SET_2M_PHY);会死机重启

hhy-ccj, 2021-03-25:
> 做client的时候调用set_connection_data_phy(CONN_SET_2M_PHY, CONN_SET_2M_PHY);会死机重启

### 解决方法
-------

> 只适用于以下版本：080 SDK

- AC636系列

   请替换到 **_include_lib\liba\br25\btctrler.a_**，并用 CodeBlocks `rebuild` 

   点击下载：[br25_080SDK_btctrler.zip](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/6204970/br25_080SDK_btctrler.zip)

hhy-ccj, 2021-04-01:
-  ### 即将发布的`090 SDK`已修复该问题

closed 2021-04-01 by hhy-ccj