Jieli-Tech/fw-AC63_BT_SDK #65: AC6351D  USB休眠唤醒功能在“AC630N_bt_data_transfer_sdk_release_v0.8.0”版本上无法实现

pepsi1021, 2021-03-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rixin-Tech, 2021-04-01:
[usb_suspend.zip](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/6241567/usb_suspend.zip)

1、include_lib/driver/device/usb/device/usb_stack.h
`#define     USB_SUSPEND_RESUME      1`

2、对应的board.c里面
```
/************************** otg data****************************/
#if TCFG_OTG_MODE
struct otg_dev_data otg_data = {
    .usb_dev_en = 0,
    .slave_online_cnt = TCFG_OTG_SLAVE_ONLINE_CNT,
    .slave_offline_cnt = TCFG_OTG_SLAVE_OFFLINE_CNT,
    .host_online_cnt = TCFG_OTG_HOST_ONLINE_CNT,
    .host_offline_cnt = TCFG_OTG_HOST_OFFLINE_CNT,
    .detect_mode = TCFG_OTG_MODE,
    .detect_time_interval = TCFG_OTG_DET_INTERVAL,
};

#endif
REGISTER_DEVICES(device_table) = {
    { "otg",     &usb_dev_ops, (void *) &otg_data},
};
```
[image]
[image]

rixin-Tech, 2021-04-06:
[696_cpu.zip](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/6261535/696_cpu.zip)

closed 2021-04-06 by rixin-Tech

rixin-Tech, 2021-07-02:
对部分场景不需要系统进入低功耗模式，没配置唤醒io为dp的用户，在mac电脑上需要打开sof中断

1、注释sof中断
    
[image]
2、收到休眠消息，使能sof中断
[image]
3、成功休眠后sof中断就会停止进入，唤醒成功后，sof中断会再次进入
[image]

rixin-Tech, 2021-11-27:
假休眠不需要打开USB_ROOT2 USB_SUSPEND_RESUME
1、把sConfigDescriptor描述符的power属性 0x80-->0xA0
[image]

2、usb_isr里面最前面标记退出休眠
3、打开suspend中断
4、在suspend中断里面记录已经进入休眠

[image]

5、在需要唤醒的地方调用一下 usb_remote_wakeup