Jieli-Tech/fw-AC63_BT_SDK #66: 632N 替换策略

Tsai1005, 2021-03-29:
|P/N|Package|Flash|I/O|Pin2Pin|替换型号|应用场景|
|:-------:|:--------:|:-------:|:-------:|:-------:|:-------:|:-------:|
|AC6321A|QFN32|2M/4M|25|No|AC6311A/AC6366C|通用蓝牙|
|AC6323A|QFN20|2M/4M|17|No|AC6313A/AC6363B|鼠标|
|AC6329F|SOP16|2M/4M|11|AC6369F|AC6369F|灯控|
|AC6329C|SOP16|2M/4M|10|AC6369C|AC6369C|灯控|
|AC6328A|SOP8|2M/4M|3|AC6368A|AC6368A|HID 类设备|
|AC6328B|SOP8|2M|2|AC6368B|AC6368B|适配器|

xjtuecho, 2021-05-26:
刚刚看发布会发布了AC632N；也看到发布了AC632N的datasheet，请问AC6328A和AC6368A主要区别是什么？

Tsai1005, 2021-05-27:
Pin的角度看基本一致，该封装没有体现出规格的差异，仅仅RAM 空间加大了，工艺改良了

xjtuecho, 2021-05-28:
> Pin的角度看基本一致，该封装没有体现出规格的差异，仅仅RAM 空间加大了，工艺改良了

AC6328A和AC6368A固件是否可以直接兼容？

Tsai1005, 2021-05-31:
程序开发角度可以兼容，SDK 已经预留了board 的差异，可以实现多个芯片实现相同的应用，开发的应用基本和SoC 不相关，但是要选择对应的board 进行配置，才能实现兼容。

Tsai1005, 2021-08-30:
632 替代631 系列