Jieli-Tech/fw-AC63_BT_SDK #67: mesh通讯速度问题

xxyang20, 2021-04-02:
vendor_client 设备发送一个开灯的消息给server设备，server设备收到消息后回复led状态给client 设备，这个过程需要200ms左右。通讯速度有什么办法可以提升吗？

hhy-ccj, 2021-04-02:
> vendor_client 设备发送一个开灯的消息给server设备，server设备收到消息后回复led状态给client 设备，这个过程需要200ms左右。通讯速度有什么办法可以提升吗？

### 解决方法
------------------------
修改 _**mesh_config_common.c**_，将节点发送 **msg** 的 **interval** 和 **duration** 缩短如下
```C
/**
 * @brief Config adv bearer hardware param when node send messages
 */
/*-----------------------------------------------------------*/
const u16 config_bt_mesh_node_msg_adv_interval = ADV_SCAN_UNIT(5); // unit: ms
const u16 config_bt_mesh_node_msg_adv_duration = 50; // unit: ms
```

xxyang20, 2021-04-06:
1、节点发送 msg 的 interval 和 duration 缩短可以提升通讯速度。修改为50/5，通讯时间100ms多一点；修改为30/3，通讯时间100ms左右变化不大，修改为20/2，测试时程序会复位，工作不正常。
2、通讯速度提升除减小 interval 和 duration外，还有什么方式？通讯速度能提升10倍吗？

hhy-ccj, 2021-04-07:
> 1、节点发送 msg 的 interval 和 duration 缩短可以提升通讯速度。修改为50/5，通讯时间100ms多一点；修改为30/3，通讯时间100ms左右变化不大，修改为20/2，测试时程序会复位，工作不正常。
> 2、通讯速度提升除减小 interval 和 duration外，还有什么方式？通讯速度能提升10倍吗？

- 080 SDK 没发现有复位情况；
- mesh 节点通讯是建立在非连接状态下，而且还涉及ack通讯，通讯时间不能保证，干扰情况下还可能出现接收延时或者接收不到的情况。

closed 2021-05-08 by Tsai1005