Jieli-Tech/fw-AC63_BT_SDK #78: [DoS] Security Vulnerability in AC6366C/AC6905X/AC6925C Bluetooth Classic Implementation (Jieli Contact needed)

Matheus-Garbelini, 2021-05-11:
**Describe the bug**
I'd like to contact Jieli team to report an AC6366C security vulnerability in Bluetooth Classic implementation.
After the attack, the device enters in a deadlock state and requires the user to manually power off/ power on the board again.

**To Reproduce**
I can send the Exploit tool via email.

**Expected behavior**
Device should not enter in deadlock state

**Screenshots**
I can share Proof of concept tool and screenshoots via email.
I cannot share any more information publicly because attackers could try to use this vulnerability in the wild to shutdown/crash products using Jieli chipsets

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Browser Chrome
 - Version 89

**Smartphone (please complete the following information):**
 Not Applicable

**Additional context**
This is my attempt to report a security vulnerability to a range of Jieli Bluetooth products, so please feel free to share some email to which I can send the report. Today, it's possible to either crash or deadlock Jieli Bluetooth SoCs/Modules.

Matheus-Garbelini, 2021-06-25:
Any updates on this?

jamin3702, 2021-09-18:
commit  "fix Security Vulnerability in AC6366C issue #78" , thanks

Matheus-Garbelini, 2021-09-18:
@jamin3702 thanks

closed 2021-09-18 by Matheus-Garbelini