Jieli-Tech/fw-AC63_BT_SDK #83: AC636x内部是否有UID或者其它加密手段？

xjtuecho, 2021-05-20:
UID可以用来做序列号，做加密，SDK中没有发现有UID的描述。或者有没有其它方案保证固件安全？

Tsai1005, 2021-05-20:
关于安全方面：
* 生成的固件会使用 -key 文件进行加密，保护代码和资源文件
* 内置Flash 使得具备一定的物理隔绝功能，防止被读取

xjtuecho, 2021-05-20:
> 关于安全方面：
> 
> * 生成的固件会使用 -key 文件进行加密，保护代码和资源文件
> * 内置Flash 使得具备一定的物理隔绝功能，防止被读取

目前我下载固件用的是代理商的key，是否意味着固件对代理商来说很容易获取？
内置SPI FLASH如果能用调试器读取的话，对增加安全性并没有意义。UID除了安全，还可以用来做序列号，区分不同的设备，厂家固化的UID更能够保证唯一性和不可修改性，无论用来做序列号还是加密都是很有用处的。

rixin-Tech, 2021-05-24:
`extern __attribute__((weak)) u8 *get_norflash_uuid(void);
u8 flash_uuid[16] = {0};
memcpy(flash_uuid, get_norflash_uuid(), 16);`

rixin-Tech, 2021-05-24:
还有一种加密二级秘钥加密模式。

Tsai1005, 2021-05-24:
> > 关于安全方面：
> > 
> > * 生成的固件会使用 -key 文件进行加密，保护代码和资源文件
> > * 内置Flash 使得具备一定的物理隔绝功能，防止被读取
> 
> 目前我下载固件用的是代理商的key，是否意味着固件对代理商来说很容易获取？
> 内置SPI FLASH如果能用调试器读取的话，对增加安全性并没有意义。UID除了安全，还可以用来做序列号，区分不同的设备，厂家固化的UID更能够保证唯一性和不可修改性，无论用来做序列号还是加密都是很有用处的。

* 根据分销模式，代理商为芯片供应方，获得key1 保护其安全性，至于其下的开发者，二次开发固件，可以由 key1 + key2 保护其安全性
* 市面上不存在读取内置Flash 的调试器

closed 2021-05-31 by Tsai1005

xjtuecho, 2021-06-01:
> * 根据分销模式，代理商为芯片供应方，获得key1 保护其安全性，至于其下的开发者，二次开发固件，可以由 key1 + key2 保护其安全性
> * 市面上不存在读取内置Flash 的调试器

> `extern __attribute__((weak)) u8 *get_norflash_uuid(void); u8 flash_uuid[16] = {0}; memcpy(flash_uuid, get_norflash_uuid(), 16);`

这个有用，准备试一下，谢谢。内部SPI FLASH读取功能已经做好了。

xjtuecho, 2021-06-02:
> `extern __attribute__((weak)) u8 *get_norflash_uuid(void); u8 flash_uuid[16] = {0}; memcpy(flash_uuid, get_norflash_uuid(), 16);`

使用get_norflash_uuid(void)这个API读取16个字节，试了5片AC6368A，结果分别如下：
```
112002270404062500A30173FFFFFFFF
112002270404071501C300E6FFFFFFFF
112002270404062500A60174FFFFFFFF
112002270404071501BB00E6FFFFFFFF
1120022704040715006E0157FFFFFFFF
```
从结果来看，128位是不同的，但是最后32位都是0xFF，请帮忙检查get_norflash_uuid(void)函数是否返回了完整的16字节，如过是的话，就是正常的。