Jieli-Tech/fw-AC63_BT_SDK #85: 关于6369一些特性能否完善？

lewisxhe, 2021-05-23:
1. 对于新手用户，这个sdk不算友好，直接选定6369 能顺利编译过去，但是无法顺利下载，因为6369才256kb flash，编译的代码量完全超出可用空间，必须得手动裁剪
2. 实际examples 寥寥无几 ，能用基本为 0，无寄存器手册，必须自己手动去扒代码，浪费客户时间
3. 我在代码中看到 ir key 的相关代码，但是在6369中并没有实际的文件，只有头文件，我在配置中看到已经有内置led灯驱动，无论是单io 还是双io都无法顺利点亮，而且636x根本没有音频，但是配置里很多相关音频部分 ，代码冗余(ps:我能猜出代码是祖传代码 copy过来的 )
4. 技术支持不够友好，无论代理还是社区，还是github ，谁用谁知道 。

rixin-Tech, 2021-05-24:
2 3两点正在参考其他通用mcu写example。

Tsai1005, 2021-05-24:
* 关于256K Flash 容量，可以理解为极具性价比的定制方案，考虑到SDK 作为功能展示的版本（包含调试信息），必然不能直接作为生产代码，不过，可以通过增加`功能优化配置说明`和`已经优化好256K 的例子`作为参考，解决这个问题。
* examples 均以市场调研结果进行迭代，examples 和 芯片在不断迭代增加，至于能用为'0' 则需要看具体产品功能，如果有通用性，可以纳入后续examples 考虑。另外，用户手册（寄存器说明）可以联系代理商获得。
* 63 系列是可以同时支持63、69 系列的，所以会有音频部分配置，归因于数传市场的多样性。
* 技术支持的不够友好，可以量化一下是哪些问题？后续引入深圳FAE 也可以有针对性解决。

lewisxhe, 2021-05-24:
@Tsai1005 
1. 性价比归性价比,价钱只是芯片的一部分，你不能用低价来提高客户对芯片编程的成本，我是第一次使用杰理的芯片，我之前使用@espressif 的芯片，他们家的编程体验非常的高，基本不需要找原厂支持.   生产代码 ？ 这可能吗 ？ 我需要一些通用的例子或者有完善的资料来对芯片进行编程，而不是直接将你们代码安排在生产上 .
2.  examples 请参考 @espressif 或者其他开源芯片产商的路线
3.  这点我可以同意
4. 追根揭底是太过闭源，什么东西都需要找代理获得，有些代理技术水平本身有限，而且服务范围大，并不能获得好的体验，有些问题问个几天都没能解决，你说寄存器手册能通过 代理获得，但是我我得到的答案并不是的，他们说寄存器手册他们也没有 ！

Tsai1005, 2021-05-25:
* 1 - 生产代码，即公模程序，市场对这个会有一定需求，也是我们现阶段考虑到的，这样的好处是对没有开发能力的代理商友好，至于有一定开发需求的开发者体验，还在优化当中，你提的几点建议都很好。
* 2 - 可以考虑添加更有代表性的demo，方便派生出接近用户需求的代码。具体可以发邮件到`caibingquan@zh-jieli.com` 详细沟通
* 4 - 我们会考虑开放用户手册（常规半导体厂商做法），具体方法后续会公布在github 上。

lewisxhe, 2021-05-25:
好，现在抛开其它的 ，谈谈6369F这颗芯片.

现在我遇到了以下几个问题:

1. RTC 不知道如何使用rtc_alarm ，如果直接在0.9.2 SDK中直接开启
```
#define TCFG_RTC_ALARM_ENABLE           1
```

将会收获这些没有定义的错误

```
F:/JL/JL_AC6963F/apps/JL_AC6963F/rtc_alarm.c:157:9: error: implicit declaration of function 'set_alarm_ctrl' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        set_alarm_ctrl(0);
        ^
F:/JL/JL_AC6963F/apps/JL_AC6963F/rtc_alarm.c:171:30: warning: declaration of 'struct rtc_dev_platform_data' will not be visible outside of
      this function [-Wvisibility]
void alarm_init(const struct rtc_dev_platform_data *arg)
                             ^
F:/JL/JL_AC6963F/apps/JL_AC6963F/rtc_alarm.c:171:6: error: conflicting types for 'alarm_init'
void alarm_init(const struct rtc_dev_platform_data *arg)
     ^
F:/JL/JL_AC6963F/apps/JL_AC6963F/include\rtc_alarm.h:41:6: note: previous declaration is here
void alarm_init(const struct rtc_dev_platform_data *arg);
     ^
F:/JL/JL_AC6963F/apps/JL_AC6963F/rtc_alarm.c:173:5: error: implicit declaration of function 'rtc_init' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
    rtc_init(arg);
```

如果将其他芯片的rtc.h 拷贝放在br25上,将会收获链接错误,因为我不知道sdk中br25 是否实现了rtc的功能

```
F:/JL/JL_AC6963F/cpu/br25/tools\sdk.elf.o: In function `alarm_calculate_time_by_week_mode':
F:/JL/JL_AC6963F/apps/JL_AC6963F/rtc_alarm.c:99: undefined reference to `caculate_weekday_by_time'
F:/JL/JL_AC6963F/cpu/br25/tools\sdk.elf.o: In function `alarm_update_all_time':
F:/JL/JL_AC6963F/apps/JL_AC6963F/rtc_alarm.c:141: undefined reference to `rtc_ioctl'
F:/JL/JL_AC6963F/apps/JL_AC6963F/rtc_alarm.c:161: undefined reference to `rtc_ioctl'
F:/JL/JL_AC6963F/rule.mk:35: recipe for target 'out' failed
make[1]: *** [out] Error 1
make[1]: Leaving directory 'F:/JL/JL_AC6963F/apps'
Makefile:87: recipe for target 'all' failed
make: *** [all] Error 1
```

2. 我无法得知如何将用户信息存储在flash内,尝试调用`syscfg_read`和`syscfg_write`,虽然返回值正常,但是重启过后还是没有保存,查看相关issue之后,
https://github.com/Jieli-Tech/fw-AC63_BT_SDK/issues/86 按照@rixin-Tech 说的照做,出现问题 https://github.com/Jieli-Tech/fw-AC63_BT_SDK/issues/86#issuecomment-847722396

3. 关于PWMLED,直接在配置里进行更改使能,led灯不亮

```
#define TCFG_PWMLED_ENABLE                  1                           //是否支持PMW LED推灯模块
#define TCFG_PWMLED_IOMODE                  LED_ONE_IO_MODE             //LED模式，单IO还是两个IO推灯
#define TCFG_PWMLED_PIN                     IO_PORTB_06                 //LED使用的IO口
```

尝试调用开启和设定模式,还是无法工作
```
   pwm_led_init(&pwm_led_data);
   pwm_led_mode_set(PWM_LED_ALL_ON);
   pwm_led_set_on();
```

4. 关于MCPWM,直接初始化后,调满占空比,灯闪的不停,示波器查看在每个周期内都有几个ms的下降沿

```
    timer_pwm_init(PWM0_TIMER, 1000, 10000, JL_TIMER5, IO_PORTB_07);
    timer_pwm_init(PWM1_TIMER, 1000, 10000, JL_TIMER3, IO_PORTB_05);
```

rixin-Tech, 2021-05-26:
1、636 芯片芯片没RTC模块

closed 2021-05-28 by lewisxhe