Jieli-Tech/fw-AC63_BT_SDK #86: SDK是否可以提供内部FLASH的P/E相关的API?

xjtuecho, 2021-05-24:
AC636x系列内置的256kB FLASH，读取可以直接在CPU地址0x1000000读取，擦除和写入没有找到相关的API，这些API在syscfg_write()这样的函数中肯定是有用到的，是否可以提供接口给用户使用呢？客户实现自己的参数存储，固件升级功能都是需要使用这个API的。

rixin-Tech, 2021-05-24:
[flash读写demo.zip](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/6529351/flash.demo.zip)

xjtuecho, 2021-05-24:
> [flash读写demo.zip](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/6529351/flash.demo.zip)

```
typedef enum _FLASH_ERASER {
    CHIP_ERASER,
    BLOCK_ERASER,//4k
    SECTOR_ERASER,//64k
} FLASH_ERASER;
```
这里BLOCK_ERASER和SECTOR_ERASER的注释是不是反了？按照0xEB6012这颗NOR FLASH的资料，它是支持256字节的Page Erase的，而且从VM的起始地址来看，并没有对齐到4kB，只是对齐到了256B，应该是支持256B的Page Erase吧？

rixin-Tech, 2021-05-24:
`typedef enum _FLASH_ERASER {
    CHIP_ERASER,
    BLOCK_ERASER,//64k
    SECTOR_ERASER,//4k
PAGE_ERASER,//256
} FLASH_ERASER;`

不同批次内置flash 不保证都支持page eraser。
后续新的sdk，擦除的接口会修改成
`extern bool sfc_erase_zone(u32 addr, u32 len);`
内部处理page 或者sector 两种擦除情况。

xjtuecho, 2021-05-24:
> `typedef enum _FLASH_ERASER { CHIP_ERASER, BLOCK_ERASER,//64k SECTOR_ERASER,//4k PAGE_ERASER,//256 } FLASH_ERASER;`
> 
> 不同批次内置flash 不保证都支持page eraser。
> 后续新的sdk，擦除的接口会修改成
> `extern bool sfc_erase_zone(u32 addr, u32 len);`
> 内部处理page 或者sector 两种擦除情况。

我这边初步测试，fopen直接返回了NULL。原因还在查找中，芯片为AC6368A。
Demo中定义了下面的2个宏，芯片实际容量为256kB，这里定义了1MB？是否会有问题？
```
#define CONFIG_SDFILE_ENABLE
#define CONFIG_FLASH_SIZE       (1024 * 1024)
```
另外，Demo中`USER_FILE_NAME`这个宏实际展开后结果是什么呢？它依赖`USE_SDFILE_NEW`和`VFS_ENABLE`这两个宏。

xjtuecho, 2021-05-24:
> `typedef enum _FLASH_ERASER { CHIP_ERASER, BLOCK_ERASER,//64k SECTOR_ERASER,//4k PAGE_ERASER,//256 } FLASH_ERASER;`
> 
> 不同批次内置flash 不保证都支持page eraser。
> 后续新的sdk，擦除的接口会修改成
> `extern bool sfc_erase_zone(u32 addr, u32 len);`
> 内部处理page 或者sector 两种擦除情况。
在SDK中找到了几个没有文档的API，包括src_write，sfc_read，是否可以绕过VFS直接通过这几个API访问FLASH呢？

···
typedef enum _FLASH_ERASER {
    CHIP_ERASER,
    BLOCK_ERASER,
    SECTOR_ERASER,
} FLASH_ERASER;
#if (USE_VM_API_SEL == VM_API_AC692X)
#include "flash_api.h"
#elif (USE_VM_API_SEL == VM_API_AC693X)
extern bool sfc_erase(FLASH_ERASER cmd, u32 addr);
extern u32 sfc_write(const u8 *buf, u32 addr, u32 len);
extern u32 sfc_read(u8 *buf, u32 addr, u32 len);
···

xjtuecho, 2021-05-24:
通过`sfc_read()`可以读出NOR FLASH的内容了，不过`sfc_erase()`行为看起来不太正常，通常FLASH擦除以后全部内容都是FF，实际测试擦除以后数据确实变了，但不全是FF。同时sfc_erase返回值为1，是否说明擦除成功了？
```
startAddr=0x000000 cpuAddr=0x01000000 flashAddr=0xFF201000
0x00000000 2A CF 50 1A 20 01 E0 01 8C BC 02 00 83 FF 00 00 *.P. ...........
0x00000010 61 70 70 5F 61 72 65 61 5F 68 65 61 64 00 FF FF app_area_head...
0x00000020 DD 03 49 8F 20 01 00 00 0C B8 02 00 82 FF 00 00 ..I. ...........
0x00000030 61 70 70 2E 62 69 6E 00 FF FF FF FF FF FF FF FF app.bin.........
0x00000040 E4 15 C6 68 2C B9 02 00 60 03 00 00 82 FF 00 00 ...h,...`.......
0x00000050 63 66 67 5F 74 6F 6F 6C 2E 62 69 6E 00 FF FF FF cfg_tool.bin....
0x00000060 1B 23 FF FF 00 52 03 00 00 9D 00 00 12 81 00 00 .#...R..........
0x00000070 56 4D 00 FF FF FF FF FF FF FF FF FF 00 20 00 00 VM........... ..
0x00000080 89 E0 FF FF 00 00 00 00 00 52 03 00 92 82 00 00 .........R......
0x00000090 50 52 43 54 00 FF FF FF FF FF FF FF 43 4F 44 45 PRCT........CODE
0x000000A0 53 92 FF FF 00 EF 03 00 00 10 00 00 92 81 01 00 S...............
0x000000B0 42 54 49 46 00 FF FF FF FF FF FF FF 41 55 54 4F BTIF........AUTO
0x000000C0 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
0x000000D0 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
0x000000E0 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
0x000000F0 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ................
```

rixin-Tech, 2021-05-24:
sfc_erase 在支持page擦除的ic是可以支持的。
所有的操作都需要基于fopen之后获取的flash地址，大小不能超过文件大小。
对其他地址操作是未定义行为，可能会异常，不允许这样操作。
fopen失败了，先检查一下ini文件是否配做正确，在buildlog里面会显示你新加的可读写区域起始地址已经大小信息，在sdk里面可以和这个log对比检查是否一致。

xjtuecho, 2021-05-24:
> sfc_erase 在支持page擦除的ic是可以支持的。
> 所有的操作都需要基于fopen之后获取的flash地址，大小不能超过文件大小。
> 对其他地址操作是未定义行为，可能会异常，不允许这样操作。
> fopen失败了，先检查一下ini文件是否配做正确，在buildlog里面会显示你新加的可读写区域起始地址已经大小信息，在sdk里面可以和这个log对比检查是否一致。

尝试了下ini文件配置
```
USERIF_ADR=AUTO;
USERIF_LEN=0x100
USERIF_OPT=1
```
下载时的空间分配为
```
USERIF_RESERVED_START : 0x3fe00
USERIF_RESERVED_END : 0x3ff00  
```
可以看出USERIF的起始地址为0x3FE00，代码中通过fopen，fget_attrs，sdfile_cpu_addr2flash_addr三个函数依次处理，得到的FLASH地址还是0x3FE00，感觉像是绕了一大圈又回来了。
AC6368A的16MB NOR FLASH空间为0x1000000-0x2000000，对于AC6368A，只有256kB，因此空间为0x1000000-0x1040000，通过sdfile_cpu_addr2flash_addr直接从CPU地址计算出FLASH地址是否可以？

rixin-Tech, 2021-05-24:
不行，如果不添加一个文件到ini里面，系统会把flash剩余空间全部分配给vm使用。之所以绕一圈，就是为了保障每个区域不重叠，不需要程序员维护一个map小心的处理各种地址。

xjtuecho, 2021-05-24:
> 不行，如果不添加一个文件到ini里面，系统会把flash剩余空间全部分配给vm使用。之所以绕一圈，就是为了保障每个区域不重叠，不需要程序员维护一个map小心的处理各种地址。

FLASH的0x00000-0x40000这256kB地址空间是直接映射到CPU的地址空间0x1000000-0x1040000吗？

rixin-Tech, 2021-05-24:
映射到 0x1E00000开始的位置。

xjtuecho, 2021-05-24:
> 映射到 0x1E00000开始的位置。

OK，现在完全对上了。Page Erase 测试正常。
FLASH地址：0x00000-0x40000
CPU映射到：0x1E00000-0x1E40000
FLASH的API使用的地址为：0x01000-0x41000，向后偏移了4kB
0x1000000开始大部分地址也是FLASH，后面一部分有不同。

lewisxhe, 2021-05-25:
@rixin-Tech 我使用6369F ， 我按照doc文档那样添加上，在buid时候提示错误
```
SPI nor flash online.
Online flash id: eb6012
Online flash size: 256K
Erase Falsh Size is 256
错误：预留配置的file.bin文件不存在，请检查
错误：fw文件中已经存在要添加的文件.

错误：fw文件中已经存在要添加的文件.

生成UFW文件 jl_isd.ufw 成功
已复制         1 个文件。
make[1]: Leaving directory 'F:/JL/JL_AC6963F/apps'

```

我也尝试手动创建一个空的file.bin,但是还是得到同样的错误.

rixin-Tech, 2021-05-26:
file.bin 是初始值，文件大小需要和设置的区域大小一致，如果没初始值就注释这一行

lewisxhe, 2021-05-26:
@rixin-Tech  感谢,可以正常使用了

lewisxhe, 2021-05-27:
@rixin-Tech 谢谢 可以了

closed 2021-05-31 by Tsai1005

lewisxhe, 2021-06-18:
@rixin-Tech  在6369C中我开启了 OTA之后，再操作flash就每次都失败了
```
[SDFILE]Open 0x1e10cd8, [/USERIF] Fail! (>_<)
```

ini文件如下:
```

[RESERVED_CONFIG]
BTIF_ADR=AUTO;
BTIF_LEN=0x1000;
BTIF_OPT=1;

#WTIF_ADR=BEGIN_END;
#WTIF_LEN=0x1000;
#WTIF_OPT=1;

PRCT_ADR=0;
PRCT_LEN=CODE_LEN;
PRCT_OPT=2;

VM_ADR=0;
VM_LEN=8K;
VM_OPT=1;

USERIF_ADR=AUTO;
USERIF_LEN=0x100;
USERIF_OPT=1;
```

rixin-Tech, 2021-07-19:
第一版，生成的程序很小，只需要256k flash就可以存储，此时用户区域在256k的尾巴
第二版程序变大了，256k flash存储不了，此时用户在512k的尾巴，从第一版升级到第二版的过程会导致用户数据丢失。

所以在配置用户自定义区域的时候add设置为立即数，USERIF_ADD =FLASH SIZE - USERIF_LEN - 4k