Jieli-Tech/fw-AC63_BT_SDK #87: 支持浮点数打印

rixin-Tech, 2021-05-25:


rixin-Tech, 2021-05-25:
[put_float.c.txt](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/6536220/put_float.c.txt)

xjtuecho, 2021-05-25:
借楼问一下：如何将AC6368A的printf函数重定向到USBDP管脚？代码中没有找到fputc()函数定义，printf函数在debug.c文件中定义为一个空函数，此外在maskrom里面也有定义，不过注释掉了。
```
  print = ABSOLUTE(0x117fdc);
  sprintf = ABSOLUTE(0x117fe0);
  /* printf = ABSOLUTE(0x117fe4); */
  puts = ABSOLUTE(0x117fe8);
```
如前我打印时用sprintf打印到一个buffer，然后再用串口函数打印，能用但是不太方便。如果可以直接用printf打印那还是能方便不少。

rixin-Tech, 2021-05-25:
你是重写了我们的sdk吗？
printf实现在rom里面,实现putchar，然后通过mask_init初始化一下。
`void mask_init(void *f, void *m, void *stack_free, void *stack_malloc,
               void *suppress_tick_and_sleep,
               void *StartFirstTask,
               void *exp_hook, void *i_hook,                            
               void *pchar, void *log_e)
`

xjtuecho, 2021-05-25:
> 你是重写了我们的sdk吗？
> printf实现在rom里面,实现putchar，然后通过mask_init初始化一下。
> `void mask_init(void *f, void *m, void *stack_free, void *stack_malloc, void *suppress_tick_and_sleep, void *StartFirstTask, void *exp_hook, void *i_hook, void *pchar, void *log_e) `

没有重写SDK，不过改了很多，把AT指令和LOG部分都去掉了，自己实现了命令行终端，调试更方便一些。
我有自己的printf方案，想着如果你们做了我自己实现一下fputc就能重定向的话就用你们的方案好了，看了这个10个void*参数的函数感觉还是用我自己的方案更方便一点。

closed 2021-06-01 by rixin-Tech

Williamzhangtw, 2021-07-27:
@rixin-Tech 能具体说明一下文件怎么用吗？ 加进工程后没效果，还需要如何配置