Jieli-Tech/fw-AC63_BT_SDK #88: 关于6369F GPIO中断api

lewisxhe, 2021-05-26:
我搜遍sdk 只找到了一个`port_wkup_interrupt_init` GPIO中断的接口,我尝试直接使用,但是芯片会不断的进入中断, GPIO口未接任何外设和元器件，代码如下:
```
static  void ir_key_irq_cb(void)
{
    printf("ir_key_irq_cb_:%d\n", gpio_read(IO_PORTB_04));
}

void ir_key_init(void)
{
    port_wkup_interrupt_init(IO_PORTB_04, 1, ir_key_irq_cb); //下降沿触发
}
```

是我调用的api 方法不对，还是不是这个api ？ 我没有手册,目前我开发仅仅是参考和搜寻sdk

lawrencejiabin, 2021-05-26:
带sniff低功耗的话，中断的IO要配成唤醒口const struct wakeup_param wk_param.port[]，中断在static void port_wakeup_callback(u8 index, u8 gpio)回调

lewisxhe, 2021-05-27:
> 带sniff低功耗的话，中断的IO要配成唤醒口const struct wakeup_param wk_param.port[]，中断在static void port_wakeup_callback(u8 index, u8 gpio)回调

Hi，lawrencejiabin 
我只想要普通的IO口中断,请问是哪个API ？

lewisxhe, 2021-05-27:
@lawrencejiabin  我尝试使用如下代码.但是还是不断的进入中断 .
```
struct port_wakeup port0 = {
	.pullup_down_enable = ENABLE,                            //配置I/O 内部上下拉是否使能
	.edge               = FALLING_EDGE,                      //唤醒方式选择,可选：上升沿\下降沿
	.attribute          = BLUETOOTH_RESUME,                  //保留参数
	.iomap              = IO_PORTB_04,                       //唤醒口选择
    .filter_enable      = ENABLE,
};

const struct sub_wakeup sub_wkup = {
	.attribute  = BLUETOOTH_RESUME,
};

const struct charge_wakeup charge_wkup = {
	.attribute  = BLUETOOTH_RESUME,
};

const struct wakeup_param wk_param = {
    .filter     = PORT_FLT_2ms,
	.port[1]    = &port0,
	.sub        = &sub_wkup,
	.charge     = &charge_wkup,
};

void __wakeup_callback(u8 index, u32 gpio)
{
    switch (index)
    {
    case 1:
        printf("__wakeup_callback:%d gpio:%d\n",index,gpio);
        break;
    default:
        break;
    }
}
void board_power_init(void)
{
    ....
    power_wakeup_init(&wk_param);
    port_edge_wkup_set_callback(__wakeup_callback);
    ....
}
```

log 打印全都是
```
__wakeup_callback:1 gpio:20
__wakeup_callback:1 gpio:20
....
```

lewisxhe, 2021-05-27:
@lawrencejiabin  已经解决，根本原因就是开启了低功耗,所以它会一直进入中断,把它关闭就好了

closed 2021-05-27 by lewisxhe