Jieli-Tech/fw-AC63_BT_SDK #99: AC692x系列关于苹果手机MTU问题

lxcxzlm, 2021-07-21:
你好！

       我在使用AC6926芯片BLE功能与苹果手机连接时，苹果手机声明MTU为185个字节。而同一个手机同一个APP在与Cypress 20706芯片连接时则声明MTU为527个字节。
       开始我认为这是APP设置的，叫APP开发人员去修改，然后APP开发人员告诉我苹果手机不能设置。都是系统默认的。

       我想请问一下，这两个芯片为什么会有这样的差别。客户想一包数据传输280个字节。这个在AC692的平台上能实现么?我使用的SDK为AC692x_SDK_release_V2.6.3 。
[image]
[image]

jamin3702, 2021-07-21:
可以尝试添加支持DLE feature，使ios发送较大的mtu，在lib_btctrler_config.c配置示例如下：
const uint64_t config_btctler_le_features = LE_ENCRYPTION | LL_FEAT_DATA_LEN_EXT;

jamin3702, 2021-07-21:
目前AC63 SDK 支持改feature

lxcxzlm, 2021-07-21:
692系列的没有办法了么？

lxcxzlm, 2021-08-03:
> 目前AC63 SDK支持改功能

请问692系列是否有办法能修改？ 我用的是AC692x_SDK_release_V2.6.3

jamin3702, 2021-08-04:
找代理支持一下

closed 2022-06-07 by JL-MinXian