Jieli-Tech/fw-AC63_BT_SDK #106: 关于低功耗sleep 唤醒的问题:  power_set_callback  (*powerdown_enter)(u8 step)

mikewen, 2021-08-18:
powerdown_enter 的 step 参数有没有说明?

sleep_enter_callback(u8  step) 里面检测 step==1 是什么意思, 和其他的 step 有什么区别?

powerdown_exit 的参数是系统的 uptime 时间吗? 单位是 us 吗? 

有办法知道 BLE 蓝牙广播开始的时间吗? (唤醒后多久)

Sunlicheng-Jieli, 2021-08-25:
powerdown执行流程如下：
low_power_sys_request->powerdown_enter(1)->push->powerdown_enter(2)->sleep->pop->powerdown_exit(usec)。
注释
1.power_enter为app注册回调，push对相关寄存器进行备份，例如gpio、clock等，app层可以在powerdown_enter(2)对寄存器进行操作，pop流程会对寄存器进行回复。
2.powerdown_exit(usec)，参数为此次系统请求低功耗时间，为预留参数。

Sunlicheng-Jieli, 2021-08-25:
目前没有办法知道唤醒后蓝牙广播开始的时间，具体需要做什么应用？

mikewen, 2021-08-25:
想在退出sleep ( powerdown_exit ) 前用 GPIO Enable DC-DC,  在 sleep_enter_callback 里用 GPIO Disable DC-DC.  降低功耗

蓝牙广播是功耗最高的时候, 希望能和 DC-DC 配合 timing

Sunlicheng-Jieli, 2021-08-25:
> 想在退出sleep ( powerdown_exit ) 前用 GPIO Enable DC-DC, 在 sleep_enter_callback 里用 GPIO Disable DC-DC. 降低功耗
> 
> 蓝牙广播是功耗最高的时候, 希望能和 DC-DC 配合 timing

sdk进入低功耗不需要disable dcdc

closed 2022-06-07 by JL-MinXian