Jieli-Tech/fw-AC63_BT_SDK #107: 256K 芯片 OTA 支持

mikewen, 2021-08-20:
256K 芯片 (AC6369F, AC6368A)  支持 OTA 吗?

main010, 2021-09-07:
codeblock编译的时候会提示当前代码支持哪一种升级，不同容量的芯片都是支持OTA的，主要还是要看当前剩余空间有多少

main010, 2021-09-07:
codeblocks会有如下提示
--------------------------- OTA UPDATE INFO ---------------------------
ISDdownload
| VM大小 = 0x7000
| !!!!!! FAIL:  测试盒经典蓝牙升级（大小=0x7f74）需要最小空间为 0x9000      
        ^^^^^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|        PASS:    测试盒串口升级（大小=0x3f14）需要最小空间为 0x0         
| !!!!!! FAIL:   测试盒BLE升级（大小=0x97cc）需要最小空间为 0xa000      
        ^^^^^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|        PASS:      SD卡升级（大小=0x4b20）需要最小空间为 0x5000      
|        PASS:      USB升级（大小=0x4bd3）需要最小空间为 0x5000      
| !!!!!! FAIL: BLE RCSP升级（大小=0xd8da）需要最小空间为 0xe000      
        ^^^^^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|        PASS:   用户UART升级（大小=0x38ba）需要最小空间为 0x4000      
| 此VM空间支持升级方式有：                            
| * 测试盒串口升级                                  
| * SD卡升级                                    
| * USB升级                                    
| * 用户UART升级

closed 2021-09-07 by main010