Jieli-Tech/fw-AC63_BT_SDK #109: 2.0.0版本SDK AC632N编译错误

alittleray, 2021-09-22:
**Describe the bug**
||=== Build: Release in AC632N_spp_and_le (compiler: Q32S LTO Compiler) ===|
C:\fw-AC63_BT_SDK-ac63_bt_sdk_release_v2.0.0\apps\spp_and_le\examples\trans_data\app_spp_and_le.c|348|error: implicit declaration of function 'trans_disconnect' is invalid in C99 [-Werror,-Wimplicit-function-declaration]|
C:\fw-AC63_BT_SDK-ac63_bt_sdk_release_v2.0.0\apps\spp_and_le\examples\trans_data\app_spp_and_le.c|348|note: did you mean 'at_send_disconnect'?|
..\..\..\..\apps\common\include\bt_common.h|86|note: 'at_send_disconnect' declared here|
||=== Build failed: 1 error(s), 0 warning(s) (0 minute(s), 5 second(s)) ===|

**To Reproduce**
1、打开br19的工程，编译就会出现这个问题

看了一下代码，应该是宏控制问题；
在app_spp_and_le.c里面应用了外部函数trans_disconnect()，需要用extern 声明，但是这个extern受TCFG_AUDIO_ENABLE控制；
而调用这个函数的地方受宏TCFG_USER_BLE_ENABLE控制，这两个宏不一定同时打开

jamin3702, 2021-09-22:
更新工具链版本到2.4.5以上，可以解决这个问题

jielicom, 2021-09-26:
-Werror,-Wimplicit-function-declaration 这个错误意思是，有函数没有声明的情况下就使用了。
这个本身是一个警告。2.4.5前的工具链默认把这个警告当做error处理。2.4.5开始取消了这个行为。

closed 2021-11-16 by alittleray