Jieli-Tech/fw-AC63_BT_SDK #113: log10函数接口计算结果错误

jason-kd, 2021-09-26:
**Describe the bug**
正在用ac6369c模组开发一款产品，要使用到log10对数函数，不过使用标准c库，或者sdk里面的库函数，计算结果都不对，这怎样解决呢？

jason-kd, 2021-09-26:
还有一个问题，就是浮点数无法打印出来。

jielicom, 2021-09-26:
1. log10具体如何不对。有没有例子
2. printf为了节省代码空间，没有处理%f等浮点相关的格式。sdk中应该有单独用于打印浮点的函数，例如put_float之类的。

jason-kd, 2021-09-26:
比如我这样定义：
（1）
        int s32dVoiceDb = 0；
	double  test = 0.0;

	test = 320 / 1024;
	s32dVoiceDb = (int)20 * log10l(test);
        printf("test: %d\n", s32dVoiceDb);
      
 打印出来是非常大的值：
      test:-2147483648

另外，put_float函数接口，我这么输出：
      double  test = 0.0;
      put_float(test );  或者 put_float(&test);  程序都会卡住，然后系统重启，是使用方法不对吗？我没有看到头文件说明。

jielicom, 2021-09-26:
首先，320, 1024都是 *整数*，所以 320/1024按照整数的运算方式的结果是 0。
所以 test = 0;
log10l（0）是nan

建议改成 test = 320.0 / 1024.0; 这样就是浮点运算。

其次，636目前没有双精度的硬件浮点，是使用软件模拟实现。所以会比较慢。如果精度满足的情况下，建议使用 float。例如：
float test = 320f / 1024.0f;
int voice = (int)20 * log10f(test);

log10 --- double -> double
log10f --- float -> float
log10l -- long double -> long double (636上,long double == double)

-----

put_float 具体是为什么会出问题，可能需要打开一下异常处理函数，看看是哪里的问题才清楚。

jason-kd, 2021-09-28:
按浮点数据类型修改后，log10f函数能正常使用，put_float功能暂时不使用，非常感谢！

closed 2021-10-08 by jason-kd