Jieli-Tech/fw-AC63_BT_SDK #121: AC6329C存储配置数据

alittleray, 2021-10-30:
你好，当前我在使用AC6329C开发产品，部分用户配置数据需要存储在芯片里，我找到了include_lib/system/device/vm.h。
在之前使用 AC692N系列开发时，可以通用vm_read和vm_write读写；
在AC63N系列的SDK中，vm_read和vm_write这个两函数被注释掉，请问如何读写VM以及定义存储项的ID？

rixin-Tech, 2021-10-30:
```c
//=================================================================================//
//                        系统配置项(VM, BTIF, cfg_bin)读写接口                    //
//接口说明:                                                                        //
//  1.输入参数                                                                     //
//      1)item_id: 配置项ID号, 由本文件统一分配;                                   //
//      2)buf: 用于存储read/write数据内容;                                         //
//      3)len: buf的长度(byte), buf长度必须大于等于read/write数据长度;             //
//  2.返回参数:                                                                    //
//      1)执行正确: 返回值等于实际上所读到的数据长度(大于0);                       //
//      2)执行错误: 返回值小于等于0, 小于0表示相关错误码;                          //
//  3.读写接口使用注意事项:                                                        //
//      1)不能在中断里调用写(write)接口;                                           //
//      2)调用本读写接口时应该习惯性判断返回值来检查read/write动作是否执行正确;    //
//=================================================================================//
int syscfg_read(u16 item_id, void *buf, u16 len);
int syscfg_write(u16 item_id, void *buf, u16 len);                                         
``````

详细使用介绍请查阅 include_lib/system/syscfg_id.h
 
https://github.com/Jieli-Tech/fw-AC63_BT_SDK/blob/20672a84254e72cb7f0633b0c44638621893d770/include_lib/system/syscfg_id.h#L56

alittleray, 2021-11-07:
非常感谢，请问有具体的错误码描述吗？比如我需要知道一个配置项是否是第一次读写。

alittleray, 2021-11-13:
good

closed 2021-11-13 by alittleray