Jieli-Tech/fw-AC63_BT_SDK #122: AC63 系列pwm输出介绍

rixin-Tech, 2021-10-30:
直接调用下面的api可以在任意GPIO输出对应频率及占空比的pwm信号
```c
/**
 * @param JL_TIMERx : JL_TIMER0/1/2/3
 * @param pwm_io : JL_PORTA_01, JL_PORTB_02,,,等等，支持任意普通IO
 * @param fre : 频率，单位Hz
 * @param duty : 初始占空比，0~10000对应0~100%
 */
void timer_pwm_init(JL_TIMER_TypeDef *JL_TIMERx, u32 pwm_io, u32 fre, u32 duty)
``````
调用示例：
```c
// PA1 输出470588Hz，占空比 20%的pwm波
timer_pwm_init(JL_TIMER2, IO_PORTA_01, 470588, 2000);
// PA1 输出484848Hz，占空比 20%的pwm波
timer_pwm_init(JL_TIMER3, IO_PORTA_02, 484848, 5000);
``````
由于timer前级后级都是整数分频器，实际上可能对应IO并不是如你所愿输出你需要的频率，此时需要改写timer_pwm_init里面timer的时钟源、分频、调整系统时钟等措施来输出准确的频率。
输出pwm涉及的时钟树如下图所示：

[image]

HSB时钟通过clk_set("sys",xxxxHz),配置，一般的lsb时钟是hsb时钟的div2，最高80MHz。实际的LSB时钟通过 clk_get("lsb")获取。

一般芯片支持如下HSB 时钟与LSB时钟：

| HSB（MHz） | LSB（MHz） |
| ----|----| 
| 24  | 12  |
| 32  |16   |
| 48  |24   |
| 64  |32   |
| 96  |48   |
|128  | 64  |

如果需要输出 470588Hz，由于timer的分频是整数的，470588*34=15999992，接近16MHz，所以需要选择给timer选择一个16MHz的时钟源，timer的时钟源、前级分频器选择列表如。‘TIMER_SSEL_RC16M ’ 是内部rc时钟，温飘比较大一般不采用。
最终只能选择lsb，通过设置hsb时钟为64M,让lsb时钟为32M，timer前级分频选 ‘TIMER_PSET_DIV_2  ’，后级分频34（TIMER->PRD），实现输出470588，

**注意：杰理全系列芯片要求timer前级分频之后的频率低于LSB/2.**

### 最终实现同时输出470588Hz，484848Hz 的代码如下。
```c
#define     TIMER_PSET_DIV_1            (0b0000<<4)
#define     TIMER_PSET_DIV_2            (0b0100<<4)
#define     TIMER_PSET_DIV_4            (0b0001<<4)
#define     TIMER_PSET_DIV_8            (0b0101<<4)
#define     TIMER_PSET_DIV_16           (0b0010<<4)
#define     TIMER_PSET_DIV_32           (0b0110<<4)
#define     TIMER_PSET_DIV_64           (0b0011<<4)
#define     TIMER_PSET_DIV_128          (0b0111<<4)
#define     TIMER_PSET_DIV_256          (0b1000<<4)
#define     TIMER_PSET_DIV_512          (0b1100<<4)
#define     TIMER_PSET_DIV_1024         (0b1001<<4)
#define     TIMER_PSET_DIV_2048         (0b1101<<4)
#define     TIMER_PSET_DIV_4096         (0b1010<<4)
#define     TIMER_PSET_DIV_8192         (0b1110<<4)
#define     TIMER_PSET_DIV_16384        (0b1011<<4)
#define     TIMER_PSET_DIV_32768        (0b1111<<4)

#define     TIMER_SSEL_LSB              (1<<10)
#define     TIMER_SSEL_OSC              (1<<10)
#define     TIMER_SSEL_HTC              (1<<10)
#define     TIMER_SSEL_LRC              (1<<10)
#define     TIMER_SSEL_RC16M            (1<<10)
#define     TIMER_SSEL_STD_12M          (1<<10)
#define     TIMER_SSEL_STD_24M          (1<<10)
#define     TIMER_SSEL_STD_48M          (1<<10)
#define     TIMER_SSEL_STD_IO_MUX_IN    (1<<10)

/**
 * @param JL_TIMERx : JL_TIMER0/1/2/3
 * @param pwm_io : JL_PORTA_01, JL_PORTB_02,,,等等，支持任意普通IO
 * @param fre : 频率，单位Hz
 * @param duty : 初始占空比，0~10000对应0~100%
 */
void timer_pwm_init(JL_TIMER_TypeDef *JL_TIMERx, u32 pwm_io, u32 fre, u32 duty)
{
    switch ((u32)JL_TIMERx) {
    case (u32)JL_TIMER0 :
        gpio_set_fun_output_port(pwm_io, FO_TMR0_PWM, 0, 1);
        break;
    case (u32)JL_TIMER1 :
        gpio_set_fun_output_port(pwm_io, FO_TMR1_PWM, 0, 1);
        break;
    case (u32)JL_TIMER2 :
        gpio_set_fun_output_port(pwm_io, FO_TMR2_PWM, 0, 1);
        break;
    case (u32)JL_TIMER3 :
        bit_clr_ie(IRQ_TIME3_IDX);
        gpio_set_fun_output_port(pwm_io, FO_TMR3_PWM, 0, 1);
        break;
    default:
        return;
    }
    u32 timer_src_clk = clk_get("lsb");
    //初始化timer
    JL_TIMERx->CON = 0;
    JL_TIMERx->CON |= TIMER_SSEL_LSB;                   //时钟源选择LSB时钟源
    JL_TIMERx->CON |= TIMER_PSET_DIV_2;                 //时钟源再2分频                            
    JL_TIMERx->CNT = 0;                                 //清计数值
    JL_TIMERx->PRD = timer_src_clk / 2 / fre;           //设置周期
    JL_TIMERx->PWM = (JL_TIMERx->PRD * duty) / 10000;   //设置初始占空比，0~10000对应0~100%
    JL_TIMERx->CON |= (0b01 << 0);                      //计数模式
    JL_TIMERx->CON |= BIT(8);                           //PWM使能
    //设置引脚状态
    gpio_set_die(pwm_io, 1);
    gpio_set_pull_up(pwm_io, 0);
    gpio_set_pull_down(pwm_io, 0);
    gpio_set_direction(pwm_io, 0);
                          
    printf("JL_TIMERx->PRD = 0x%x\n", JL_TIMERx->PRD);
    printf("JL_TIMERx->CON = 0x%x\n", JL_TIMERx->CON);
}                         

void timer_pwm_test(void) 
{                         
    clk_set("sys",64*1000000);
    timer_pwm_init(JL_TIMER2, IO_PORTA_01, 470588, 5000);
    timer_pwm_init(JL_TIMER3, IO_PORTA_02, 484848, 5000);
    extern void wdt_clear();
    while (1) {           
        wdt_clear();      
    }                     
}                         

``````

rixin-Tech, 2021-11-13:
[image]

因为mcpwm的时钟固定来自lsb，sdk在切换模式的时候会动态调整系统时钟，导致lsb不停变化，可以使用下面方式锁定系统时钟，保持lsb频率稳定

rixin-Tech, 2021-11-26:
[image]

mikewen, 2022-05-09:
FO_TMR0_PWM  FO_TMR1_PWM FO_TMR2_PWM FO_TMR3_PWM 在 BR25 里没有定义,  应该是什么值？

zxb-jl, 2022-12-21:
timer pwm输出频率计算表，定时时间计算表。快速计算频率
[timer pwm输出频率调节表.xls](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/10276934/timer.pwm.xls)

JOJOyh, 2025-08-01:
BR25的呢？要怎么用呢？这文档写得莫名其妙的？在群里问也没人回。