Jieli-Tech/fw-AC63_BT_SDK #128: AC6329C硬件定时器不进中断函数

alittleray, 2021-11-11:
### 芯片：AC6329C
### 代码描述：简单测试一下定时器中断，选择定时器3，时钟选24M，4分频，PRD=6000，定时器工作在计数模式。中断函数中一个变量自增，定时处理函数打印这个变量值显示进入中断次数。
### 结果：发现未进入中断函数
### 代码如下：
#define TIMERX  JL_TIMER3
#define IRQ_IDX IRQ_TIME3_IDX
static volatile u32x g_u32TimerIsCnt = 0;
/*******************************
 * 定时器中断函数
 * *****************************/
___interrupt
static void TimerIsrFunc(void)
{
    TIMERX->CON |= BIT(14);
    //++ enter interrupt times
    g_u32TimerIsCnt++;
}
/*******************************
 * 启动定时器
 * *****************************/
static void TimerStart()
{
    TIMERX->CON |= BIT(14);
    TIMERX->CON = 0;
    TIMERX->CON |= (0b110 << 10); //24M时钟
    TIMERX->CON |= (0b0001 << 4);//4分频
    TIMERX->CNT = 0;
    TIMERX->PRD = 6000;
    TIMERX->CON |= (0b01 << 0);
    request_irq(IRQ_IDX, 6, TimerIsrFunc, 0);
}

/************************************
 * 通过sys_timer_add添加，每秒执行一次
 * **********************************/
void XpfAppProc(void *lpParam)
{
    static u8x szLog[100];
    //clear watchdog
    SysExtClearWdt();
    //start timer
    static boolx bHasStartTimer = FALSEX;
    if(FALSEX == bHasStartTimer){
        bHasStartTimer = TRUEX;
        puts("start timer");
        TimerStart();
    }
    //log
    s32x s32Ret = snprintf(szLog, sizeof(szLog) - 1, "timing proc, timer isr times:%u, CON:0x%X, CNT:%u, PRD:%u\r\n", 
        g_u32TimerIsCnt, TIMERX->CON, TIMERX->CON, TIMERX->PRD);
    szLog[s32Ret] = 0;
    puts(szLog);
}

### 日志如下：
[00:00:00.793][Info]: [GATT_SERVER]adv_en:1

[00:00:00.797][Info]: [GATT_SERVER]server_work_st:1->20

[00:00:00.803][BLE_TRANS]server_state: handle=00,20
 
[00:00:00.809]init rf_osc:0 PLL_comp:0 

w[]I
PP
w[]
w[]w[]
[00:00:01.680]start timer
[00:00:01.683]timing proc, timer isr times:0, CON:0x1811, CNT:6161, PRD:6000
***************************I
PP
w[]
w[]
IPP
w[]
w[]w[]
[00:00:02.690]timing proc, timer isr times:0, CON:0x9811, CNT:38929, PRD:6000
I
PP
w[]
w[]I
PP
w[]
w[]w[]
[00:00:03.695]timing proc, timer is
r times:0, CON:0x9811, CNT:38929, PRD:6000
IPP
w[]
w[]I
PP
w[]
w[]
I
[00:00:04.695]timing proc, timer isr times:0, CON:0x9811, CNT:38929, PRD:6000
PP
w[]
w[]I
PP
w[]
w[]I

[00:00:05.695]timing proc, timer isr times:0, CON:0x9811, CNT:38929, PRD:6000
PP
w[]
w[]I
PP
w[]
w[]I
P

alittleray, 2021-11-11:
sdk版本2.0.0

lawrencejiabin, 2021-11-11:
request_irq(IRQ_IDX, 3, TimerIsrFunc, 0);，只支持到3

alittleray, 2021-11-13:
调整优先级

closed 2021-11-13 by alittleray