Jieli-Tech/fw-AC63_BT_SDK #132: 设置互补PWM为中心对齐无波形输出

alittleray, 2021-11-15:
### 芯片:AC6329C
### SDK版本:2.0.0
### 描述：使用mcpwm.c中的api输出互补pwm，设置为边沿对其正常输出pwm，设置为中心对齐无波形输出。代码如下：
```
        struct pwm_platform_data pwm_p_data;
        pwm_p_data.pwm_aligned_mode = pwm_center_aligned;         //边沿对齐
        pwm_p_data.pwm_ch_num = pwm_ch0;                        //通道号
        pwm_p_data.frequency = 1000;                            //1KHz
        pwm_p_data.duty = 7000;                                 //占空比50%
        pwm_p_data.h_pin = IO_PORTA_07;                         //任意引脚
        pwm_p_data.l_pin = IO_PORTA_08;                          //任意引脚,不需要就填-1
        pwm_p_data.complementary_en = 1;                        //两个引脚的波形, 0: 同步,  1: 互补，互补波形的占空比体现在H引脚上
        mcpwm_init(&pwm_p_data);
```

wjjjb123, 2021-11-15:
在设置占空比的函数那里，针对0占空比和满占空比采用硬件控制，后期 统一改为了左对齐。
如果左对齐能解决问题的，建议使用左对齐。如果一定要用中心对齐，修改设置占空比函数里的timer_reg->tmr_con |= 0b01;改为 timer_reg->tmr_con |= 0b10;

alittleray, 2021-11-15:
互补pwm如何加上死区控制

alittleray, 2021-11-15:
确认可行

closed 2021-11-15 by alittleray