Jieli-Tech/fw-AC63_BT_SDK #133: MCPWM死区控制问题

alittleray, 2021-11-15:
### 芯片:AC6329C
### sdk版本：2.0.0
### 描述：使用mcpwm.c中api输出频率1K占空比50%的互补方波
```
struct pwm_platform_data pwm_p_data;
        pwm_p_data.pwm_aligned_mode = pwm_edge_aligned; //边沿对齐
        pwm_p_data.pwm_ch_num = pwm_ch0;                //通道号
        pwm_p_data.frequency = 1000;                    //1KHz
        pwm_p_data.duty = 5000;                         //占空比50%
        pwm_p_data.h_pin = IO_PORTA_07;                 //任意引脚
        pwm_p_data.l_pin = IO_PORTA_08;                 //任意引脚,不需要就填-1
        pwm_p_data.complementary_en = 1;                //两个引脚的波形, 0: 同步,  1: 互补，互补波形的占空比体现在H引脚上
        mcpwm_init(&pwm_p_data);
```
输出结果正常，现在加上死区控制，在mcpwm.c的mcpwm_init函数内部，设置占空比(mcpwm_set_duty(arg->pwm_ch_num, arg->duty))函数之后加以下代码：
```
pwm_reg->ch_con0 |= (0b0111 << 12);
    pwm_reg->ch_con0 |= (0b00011 << 7);
    pwm_reg->ch_con0 |= BIT(6);
```
发现输出波形变成了同步而非互补

wjjjb123, 2021-11-15:
因为我们的蓝牙芯片主要用在蓝牙低功耗领域，虽然我们芯片有这个外设，但是很少有用户用到，大部分用户都不知道什么是互补的pwm波，更别说死区时间控制了。
所以我们的pwm驱动写的相对简单，一些功能没有加上去。
AC692的MCPWM
如果不开死区控制，H和L脚默认是同步输出的。
如果开了死区控制，则H和L脚默认是互补输出的。
所以 使能了 死区控制 之后， 结构体的参数 pwm_p_data.complementary_en 的意思正好相反。pwm_p_data.complementary_en = 0, 就可以。

alittleray, 2021-11-15:
确认可以，非常感谢。关闭mcwpm这块给个详细的文档吧。

closed 2021-11-17 by alittleray