Jieli-Tech/fw-AC63_BT_SDK #134: ac6329f使用mcpwm驱动LED，设置某些占空比会锁死遥控接收头的输入引脚，导致遥控信号接收不到。

jason-kd, 2021-11-17:
bug描述
 ac6329f使用mcpwm模块的3路输出驱动3路LED，在设置某些占空比会锁死遥控接收头的输入引脚，遥控接收头一直是高电平，没有检测到红外信号， 导致红外接收模块没有中断。（红外接收模块也是使用sdk里面的模块）

复现过程：
开启红外按键模块：
#define TCFG_IRKEY_ENABLE                   ENABLE_THIS_MOUDLE//DISABLE_THIS_MOUDLE//是否使能AD按键
#define TCFG_IRKEY_PORT                     IO_PORTA_00//IO_PORTA_08        //IR按键端口

mcpwm初始化：
	printf("mcpwm-init\n");
	g_astPwmData[0].pwm_aligned_mode = pwm_edge_aligned;		
	g_astPwmData[0].pwm_ch_num = pwm_ch0;						
	g_astPwmData[0].frequency = 10000;							
	g_astPwmData[0].duty = 0;									
	g_astPwmData[0].h_pin = IO_PORTA_02;						
	g_astPwmData[0].l_pin = IO_PORTA_02;//IO_PORTA_02;									
	g_astPwmData[0].complementary_en = 0;						
	mcpwm_init(&g_astPwmData[0]);
	
	g_astPwmData[1].pwm_aligned_mode = pwm_edge_aligned;		
	g_astPwmData[1].pwm_ch_num = pwm_ch1;						
	g_astPwmData[1].frequency = 10000;							
	g_astPwmData[1].duty = 0;								
	g_astPwmData[1].h_pin = IO_PORTB_06;						
	g_astPwmData[1].l_pin = IO_PORTB_06; 								
	g_astPwmData[1].complementary_en = 0;						
	mcpwm_init(&g_astPwmData[1]);

	g_astPwmData[2].pwm_aligned_mode = pwm_edge_aligned;		
	g_astPwmData[2].pwm_ch_num = pwm_ch2;						
	g_astPwmData[2].frequency = 10000;							
	g_astPwmData[2].duty = 0;								
	g_astPwmData[2].h_pin = IO_PORTB_07;						
	g_astPwmData[2].l_pin = IO_PORTB_07; 								
	g_astPwmData[2].complementary_en = 0;						
	mcpwm_init(&g_astPwmData[2]);

调整占空比：
	mcpwm_set_duty(g_astPwmData[0].pwm_ch_num, u32R_duty);
	mcpwm_set_duty(g_astPwmData[1].pwm_ch_num, u32G_duty);
	mcpwm_set_duty(g_astPwmData[2].pwm_ch_num, u32B_duty);

当设置某些占空比，比如：
	mcpwm_set_duty(g_astPwmData[0].pwm_ch_num, 8980);
	mcpwm_set_duty(g_astPwmData[1].pwm_ch_num, 8980);
	mcpwm_set_duty(g_astPwmData[2].pwm_ch_num, 8980);
设置后，pwm输出正常，但是此时，会影响到红外接收，接收引脚 IO_PORTA_00，一直是高电平，按遥控器也没有检测到信号。

wjjjb123, 2021-11-19:
按照你的配置，我们测试没有复现到。
另外建议一点，h_pin和l_pin最好不要填同一个引脚，没有的填 -1。

jason-kd, 2021-11-19:
难道环境不一样吗，我这里时必现的。而且最开始l_pin 就是设置为-1，出现问题后，我才尝试改为同一个引脚，发现也是同样的问题。

closed 2022-06-07 by JL-MinXian