Jieli-Tech/fw-AC63_BT_SDK #135: AC6329C的11脚可以作为普通GPIO使用吗

alittleray, 2021-11-22:
### 芯片：AC6329C
### 描述：想把AC6329C的11脚USB0DM作为普通GPIO使用，是不是不可以。
[image]
[image]

jamin3702, 2021-11-22:
可以的，用封装好的gpio接口配置，gpio.h； 示例
//用io反转推灯
#define TCFG_PWMLED_PIN     IO_PORT_DP
#define KEYF_LED_ON()    gpio_direction_output(TCFG_PWMLED_PIN, 1)
#define KEYF_LED_OFF()   gpio_direction_output(TCFG_PWMLED_PIN, 0)

#define KEYF_LED_INIT()  {gpio_set_dieh(TCFG_PWMLED_PIN, 0);gpio_set_die(TCFG_PWMLED_PIN, 1);gpio_set_pull_down(TCFG_PWMLED_PIN, 0);gpio_set_pull_up(TCFG_PWMLED_PIN, 0);} //

alittleray, 2021-11-22:
这个GPIO的序号是多少呢

closed 2021-11-22 by alittleray