Jieli-Tech/fw-AC63_BT_SDK #139: ac6329f模组开启闹铃模块后，闹铃模块没生效

jason-kd, 2021-12-14:
问题描述：
1. ac6329f模组配置开启闹铃：
//*********************************************************************************//
//                                  RTC_ALARM配置                                  //
//*********************************************************************************//
#define TCFG_RTC_ALARM_ENABLE               ENABLE_THIS_MOUDLE //DISABLE_THIS_MOUDLE

2. app_main主函数里开启和使用sdk的demo测试接口，测试闹铃功能。
 	alarm_rtc_start();
	user_alarm_test();

3. sdk的代码如下：
//参考的测试代码
void user_alarm_test(void)
{
    T_ALARM tmp_alarm = {0};
    rtc_ioctl(IOCTL_GET_SYS_TIME, (u32)&tmp_alarm.time);
    tmp_alarm.en = 1;                           //初始化默认打开

    printf("1-time:%d-%d-%d:%d,%d,%d\n", tmp_alarm.time.year, tmp_alarm.time.month, tmp_alarm.time.day,
    		tmp_alarm.time.hour, tmp_alarm.time.min, tmp_alarm.time.sec);
    
#if 1
    tmp_alarm.mode = E_ALARM_MODE_ONCE;         //此闹钟只起作用一次
    //tmp_alarm.time.hour = 0;
    tmp_alarm.time.min  = 1;
    alarm_add(&tmp_alarm, 0);
#endif

#if 1
    tmp_alarm.mode = E_ALARM_MODE_EVERY_DAY;    //此闹钟每天都起作用
    //tmp_alarm.time.hour = 0;
    tmp_alarm.time.min  = 2;
    alarm_add(&tmp_alarm, 1);
#endif

#if 1
    tmp_alarm.mode = E_ALARM_MODE_EVERY_MONDAY | E_ALARM_MODE_EVERY_WEDNESDAY | E_ALARM_MODE_EVERY_SATURDAY; //此闹钟周1周3周6起作用
    //tmp_alarm.time.hour = 0;
    tmp_alarm.time.min  = 3;
    alarm_add(&tmp_alarm, 2);
#endif
}

4. 上面设置1分钟的闹铃时间，但是没有看到闹铃回调的打印，一直有这个时间打印（该时间是初始化 def_sys_time是设置的时间）：
[18:45:23.566]收←◆[00:31:14.165]rtc_read_sys_time: 2021-12-14 18:0:0

[18:45:25.566]收←◆[00:31:16.165]rtc_read_sys_time: 2021-12-14 18:0:0

[18:45:27.566]收←◆[00:31:18.165]rtc_read_sys_time: 2021-12-14 18:0:0

[18:45:29.566]收←◆[00:31:20.165]rtc_read_sys_time: 2021-12-14 18:0:0

lawrencejiabin, 2021-12-14:
用的是什么版本的SDK？？？默认是需要挂32K晶振才能跑RTC的，100需要打补丁才能支持免晶振，200就不用打补丁

jason-kd, 2021-12-15:
1. sdk版本号是Rev 1.0.0—— 2021 年 6 月 11 日的版本
2. 另外我其实是想用mktime和localtime这两个标准c库接口进行时间戳的转换，但是我发现将 C:\JL\pi32\q32s-lib里面的libc.a库函数，增加到编译器里，编译还是出错，提示没有定义这两个函数。我是在编译器的setting 选项的 compiler settings选项的 linker setting增加了链接该库。

jason-kd, 2021-12-16:
我们在Rev 1.0.0版本上开发不少功能，移到200上面会比较耗时费力，给我们一个100版本上的补丁吧。

closed 2022-06-07 by JL-MinXian