Jieli-Tech/fw-AC63_BT_SDK #143: BLE定向广播显示线程错误

Orang-Lab, 2022-01-11:
1.在广播参数里设置ADV_DIRECT_IND定向广播；

[image]
2.HCI事件 定向广播超时事件
[image]
3.log的结果显示 线程报错 导致后续无蓝牙广播
[image]
4.察看了le_gatt_server.c文件，发现里面的处理机制是先产生事件通知再去设置工作状态，如果事件通知给应用层，应用层也设置新的工作状态，会导致设置不生效。

jamin3702, 2022-01-11:
确实存在上述现象，应该先更新工作状态，再通知应用，多谢指出！

jamin3702, 2022-01-11:
把顺序换一下，试试

Orang-Lab, 2022-01-11:
已换了  不行

Orang-Lab, 2022-01-11:
显示error了  麻烦帮忙看下是哪里导致的底层错误，然后才是状态设置的先后问题

jamin3702, 2022-01-11:
上传打印log 和 你的修改看看

Orang-Lab, 2022-01-11:
[00:00:09.606][Info]: [GATT_SERVER]__gatt_server_just_new_dev_adv

[00:00:09.607][Info]: [GATT_SERVER]new_dev_adv

[00:00:09.607][Info]: [GATT_SERVER]adv_en:1

[00:00:09.608][Info]: [GATT_SERVER]server_work_st:1->20

[00:00:09.609][MUL-PER]server_state: handle=00,20
 
[00:00:09.610][MUL-PER]ble_state:ff===>20
 
[00:00:09.612]init rf_osc:0 PLL_comp:0 
[00:00:09.631][Info]: [USER_MULTI]open --------------------->
[00:00:10.897][Debug]: [LL_E]HCI_SUBEVENT_LE_CONNECTION_COMPLETE 3c

[00:00:10.898](error): <Error>: [LL_ADV]TO - LL Adv High Duty Timeout

[00:00:10.899][Info]: [GATT_SERVER]DIRECT_ADV TO!

[00:00:10.900][MUL-PER]------------------------------------->>>>>>>>>>DIRECT_ADV_TIMEOUT:2 
[00:00:10.902][Info]: [GATT_SERVER]server_work_st:20->2

[00:00:10.904][MUL-PER]server_state: handle=00,02
 
[00:00:10.905][MUL-PER]ble_state:20===>02

Orang-Lab, 2022-01-11:
[image]

jamin3702, 2022-01-11:
截图没看到调整这个接口  ble_gatt_server_set_adv_config(&hogp_server_adv_config);

Orang-Lab, 2022-01-11:
[image]
mo'r默认是大家都懂了

Orang-Lab, 2022-01-11:
这样设置后  开广播使能就会出现log   error

jamin3702, 2022-01-11:
[image]
关闭这个，需要调接口开广播

Orang-Lab, 2022-01-11:
是的  这个了解  但是不是产生错误的原因吧  我是想协助解决为什么底层报错  不是纠结广播与否问题

Orang-Lab, 2022-01-11:
打开自动广播是会开了广播      但还是会报错 不是我想要了解的问题   如果说这个报错是底层本身存在 我就不纠结了
[image]

jamin3702, 2022-01-11:
这个不是底层错误，官方协议定义广播类型。高频率的定向广播只能维持1.28秒；未被连接，时间到后结束广播；spec上有描述

Orang-Lab, 2022-01-11:
但是有看到error

closed 2022-06-07 by JL-MinXian