Jieli-Tech/fw-AC63_BT_SDK #152: AC6329C Edr Emitter咨询

alittleray, 2022-03-31:
芯片：AC6329C
SDK版本：2.0.0
描述：当前使用app_spp_and_le 实例开发，但是需要使用经典蓝牙搜索周边设备，于是增加了#define EDR_EMITTER_EN 1宏定义，但是我发现搜索不到蓝牙，请指导。日志如下：
[00:20:05.825][Info]: [EDR_EM]>>>   bt_search_start

[00:20:31.441][Info]: [SPP_AND_LE]----spple_bt_hci_event_handler reason 1 0
[00:20:31.448][Info]: [COMM_EDR]--------bt_comm_edr_hci_event_handler reason 1 0
[00:20:31.456][Info]: [COMM_EDR] HCI_EVENT_INQUIRY_COMPLETE 

[00:20:31.462][Info]: [EDR_EM]bt_emitter_search_complete,469

[00:20:31.470][Info]: [EDR_EM]>>>   bt_search_start

[00:20:57.081][Info]: [SPP_AND_LE]----spple_bt_hci_event_handler reason 1 0
[00:20:57.089][Info]: [COMM_EDR]--------bt_comm_edr_hci_event_handler reason 1 0
[00:20:57.097][Info]: [COMM_EDR] HCI_EVENT_INQUIRY_COMPLETE 

[00:20:57.103][Info]: [EDR_EM]bt_emitter_search_complete,469

jamin3702, 2022-04-01:
有关了BLE吗

alittleray, 2022-04-01:
你好，关闭ble是指关闭ble广播吗

jamin3702, 2022-04-02:
是不是信号比较弱，靠近或者换个板子试试

alittleray, 2022-04-02:
应该不是的，我的电脑和手机蓝牙都是开着的，周边也有很多蓝牙，手机可以搜出一大堆。在钉钉群里技术人员建议我用app_dongle，我用2.1.0版本打开edr_emitter试了一下，也没收到。请问相关的技术资料吗

jamin3702, 2022-04-06:
用 v2.1.0 更新库测测，详见 #153

closed 2022-06-07 by JL-MinXian