Jieli-Tech/fw-AC63_BT_SDK #153: AC632N上如何使用经典蓝牙搜索功能

alittleray, 2022-03-31:
芯片：AC6329C
SDK版本：2.1.0
项目需求：
1、手机作为中心与设备通过BLE进行数传；
2、设备通过经典蓝牙搜索功能发现周边的蓝牙设备；
请问在app_spp_and_le实例下是否能够实现，该如何配置

jamin3702, 2022-04-06:
[btctrler.zip](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/8425237/btctrler.zip)
替换这个库，测测

alittleray, 2022-04-06:
感谢回复，当前项目使用的是2.0.0版本的库，替换后编译有错误，是需要升级到2.1.0吗
[image]

jamin3702, 2022-04-06:
库是对应v2.1.0版本

alittleray, 2022-04-06:
使用2.1.0版本sdk，实例app_spp_and_le，在该实例下增加了EDR_EMITTER_EN
#if CONFIG_APP_SPP_LE
//配置双模同名字，同地址
#define DOUBLE_BT_SAME_NAME                0 //同名字
#define DOUBLE_BT_SAME_MAC                 0 //同地址
#define CONFIG_APP_SPP_LE_TO_IDLE          0 //SPP_AND_LE To IDLE Use
//蓝牙BLE配置
#define CONFIG_BT_GATT_COMMON_ENABLE       1 //配置使用gatt公共模块
#define CONFIG_BT_SM_SUPPORT_ENABLE        0 //配置是否支持加密
#define CONFIG_BT_GATT_CLIENT_NUM          0 //配置主机client个数 (app not support)
#define CONFIG_BT_GATT_SERVER_NUM          1 //配置从机server个数
#define CONFIG_BT_GATT_CONNECTION_NUM      (CONFIG_BT_GATT_SERVER_NUM + CONFIG_BT_GATT_CLIENT_NUM) //配置
#define EDR_EMITTER_EN                     1 //蓝牙(edr主机)

编译还是会出现错误，是否是因为其他宏没打开或者其他库也需要一起更新，期待回复，谢谢
[image]

jamin3702, 2022-04-06:
这边试过是可以的，你sdk版本不对应，你用的应该是2.0.0

alittleray, 2022-04-07:
我下载了最新的2.1.0版本，替换之后编译没有问题。使用的是实例是CONFIG_APP_SPP_LE, 增加了打开Edr_emitter的宏，但是还是搜不到，是不能用这个实例吗。
#if CONFIG_APP_SPP_LE
//配置双模同名字，同地址
#define DOUBLE_BT_SAME_NAME                0 //同名字
#define DOUBLE_BT_SAME_MAC                 0 //同地址
#define CONFIG_APP_SPP_LE_TO_IDLE          0 //SPP_AND_LE To IDLE Use

//蓝牙BLE配置
#define CONFIG_BT_GATT_COMMON_ENABLE       1 //配置使用gatt公共模块
#define CONFIG_BT_SM_SUPPORT_ENABLE        0 //配置是否支持加密
#define CONFIG_BT_GATT_CLIENT_NUM          0 //配置主机client个数 (app not support,应用不支持使能)
#define CONFIG_BT_GATT_SERVER_NUM          1 //配置从机server个数
#define CONFIG_BT_GATT_CONNECTION_NUM      (CONFIG_BT_GATT_SERVER_NUM + CONFIG_BT_GATT_CLIENT_NUM) //配
#define EDR_EMITTER_EN                     1 //蓝牙(edr主机)

我在Edr_emitter.c每个函数都加了函数名打印，日志如下：

[00:03:00.253][Info]: [EDR_EM]bt_search_start >>>>>>>>>

[00:03:25.864][Info]: [SPP_AND_LE]----spple_bt_hci_event_handler reason 1 0
[00:03:25.871][Info]: [COMM_EDR]--------bt_comm_edr_hci_event_handler reason 1 0
[00:03:25.879][Info]: [COMM_EDR] HCI_EVENT_INQUIRY_COMPLETE 

[00:03:25.884][Info]: [EDR_EM]bt_emitter_search_complete

[00:03:25.890][Info]: [EDR_EM]bt_emitter_search_complete,474

[00:03:25.898][Info]: [EDR_EM]bt_search_start >>>>>>>>>

[00:03:51.513][Info]: [SPP_AND_LE]----spple_bt_hci_event_handler reason 1 0
[00:03:51.520][Info]: [COMM_EDR]--------bt_comm_edr_hci_event_handle
r reason 1 0
[00:03:51.528][Info]: [COMM_EDR] HCI_EVENT_INQUIRY_COMPLETE 

[00:03:51.534][Info]: [EDR_EM]bt_emitter_search_complete

[00:03:51.539][Info]: [EDR_EM]bt_emitter_search_complete,474

[00:03:51.548][Info]: [EDR_EM]bt_search_start >>>>>>>>>

jamin3702, 2022-04-07:
你搜索的对象是什么设备类型？ 搜索个hid设备或者音箱应该是可以的

alittleray, 2022-04-08:
我是需要搜索到我的手机，搜索对象类型这个是可以设置的吗

jamin3702, 2022-04-12:
例子目前支持搜索连接 HID类型设备

tpunix, 2022-04-13:
主机搜索到设备(无类型限制)后，会调用bt_emitter_search_result这个函数做处理。找到这个函数，加点打印看看是否搜索到了什么。要学会看代码

alittleray, 2022-04-17:
加了打印，愣是什么都没有搜到