Jieli-Tech/fw-AC63_BT_SDK #155: 动态改变 TCFG_LOWPOWER_LOWPOWER_SEL

mikewen, 2022-04-13:
现在编译时可以改变 
#define TCFG_LOWPOWER_LOWPOWER_SEL			SLEEP_EN                     //SNIFF状态下芯片是否进入powerdown

有没有 API 可以动态改变 SNIFF状态下芯片是否进入powerdown ?

lawrencejiabin, 2022-04-14:
https://gitee.com/jieli-tech_fae/IOT-AC630N/blob/master/doc/software/%E9%A9%B1%E5%8A%A8/%E4%BD%8E%E5%8A%9F%E8%80%97%E6%B5%81%E7%A8%8B.md

注册外设查询空闲函数

//SDK任意地方添加此段代码，全局修改此变量，底层会自动获取判断
u8 can_enter_lp = 0; //置 1 可进睡眠， 置 0 不可进睡眠
static u8 custom_idle_query(void)
{
	if(can_enter_lp){
		return 1;
	}else{
		return 0;
	}
} 
REGISTER_LP_TARGET(custom_lp_target) = {  //结构体名字也随便取个名字
	.name = "custom_lp",                  //随便取个名字，不要重复
	.is_idle = custom_idle_query,
};

jamin3702, 2022-04-14:
参考下面注册一个这样的函数回调，改变值 is_hidkey_active 来动态控制系统是否进入powerdown。

static volatile u8 is_hidkey_active = 0;//1-临界点,系统不允许进入低功耗，0-系统可以进入低功耗
//system check go sleep is ok
static u8 hidkey_app_idle_query(void)
{
    return !is_hidkey_active;
}

REGISTER_LP_TARGET(app_hidkey_lp_target) = {
    .name = "app_hidkey_deal",
    .is_idle = hidkey_app_idle_query,
};

closed 2022-06-07 by JL-MinXian