Jieli-Tech/fw-AC63_BT_SDK #176: 各位大佬，求助：BFU 文件能反编译吗？能的话，如何反编译？

XuPengfei-1020, 2022-08-21:
我遇到的问题：
我购买了一个蓝牙音响：漫步者 g2000。这个音响哪儿都挺好的，但是有一个非常糟糕的设计，它在开关机，按键切换模式的时候，都会播放一个声量巨大的女声语音提示音：（“轰~~”（跑车轰鸣声），“蓝牙模式”，“声卡输入”，“线路输入”，“电影模式”，“游戏模式”，”音乐模式“，(甜美的女声)），声音很甜美，但是也实在是太尴尬了！
----------------------------------------------------

我怎么找到这里的：
我通过搜索引擎，找到了一个官方固件，发现它的格式是 bfu 格式的：jl_690x.bfu，我通过调查，发现这个格式是贵公司的蓝牙芯片固件编译后的格式。进一步调查，我找到了贵公司的官网，发现有些工具软件文档，然后从文档中找到了这里。

求助内容(我的计划)：
所以来这里求助：BFU 格式文件可否被反编译？我想把文件反编译了，把里面的语音提示文件替换了，然后编译回去，再刷回去。
我不是做单片机开发的，我是做软件的，有一定的计算机知识基础，求官方提点一下。谢谢你们！

JL-MinXian, 2022-08-23:
没有相关的工具提供修复bfu文件

XuPengfei-1020, 2022-10-08:
> 你检查是否有外置flash，如果有，你可以尝试找到提示声音文件，替换为静音并重新上传到外置flash。 如果没有外部闪存来执行此操作，则不值得重写固件。

收到，感谢提醒。

XuPengfei-1020, 2022-10-08:
> 没有相关的工具提供修复bfu文件

没有，这个是厂家搞的，我找到 buf 文件都费了很大劲

closed 2022-10-08 by XuPengfei-1020