Jieli-Tech/fw-AC63_BT_SDK #180: how to get Complie Tool in a realistic way

bm16ton, 2022-09-03:
im filing this under bug because its clearly a mistake. The download link which is in chineadr only does t except the password when using google translate..First why post it somewhere that needs a download password? From a github page promoting opensource? (and a fair amount of closed source) when i get in by using the chinease only website everywhere i click wants me to download sum app (a 200mb+ apk on my android device or rpm/deb package of unknown size on linux machine) this is absurd! Who thought this was acceptable? Is there anyway someone can post a sane link (IE without foreign nations spyware) or email me the files? i hope the other issues written in a language i dodont know are saying the same thing and that everyone hasnr become complacent with this. If your freely giving the software away to use absolutely no reason to use such an insane hosting method, no good reason could be given for this, its motivation is clearly installing whatever that software is on as many peoples devices as possible and nothing more.

0x0fe, 2022-09-14:
You are clearly not the target audience, unless you are an engineer, working in China on mass produced designs, you just have no reason to use it.

bm16ton, 2022-09-14:
Ha nice! I guesd i sshould have known that since it said so inthe readme and didnt say things like "ooen source" ,which woulda been confusing cuz that implies open to all. But lesson to be learned if you only want a small selective group to use it just fn say so in the gd readme. Seems like a very strange way to distribute the software for those people. Like the local burger joint using the channel 6 news to say my order is ready

0x0fe, 2022-09-16:
it surely isnt open source by any standard

0x0fe, 2022-10-20:
normal, its not for public nor hobby use, no need to waste time with this guy if he couldnt even figure out how to install the toolchain.

bm16ton, 2022-10-20:
no idea why you thought i couldnt install the toolchain. The whole complaint was as stated..i guess i shouldnt waiste my time on someone who cant read the posts. PS already programmed my module and without the chinease spyware, not as bad as you made out, but as weve established reading isnt your thing.

bm16ton, 2022-10-22:
all kidding and fun mockery aside i do appreciate you responding and any and all work that goes into a project like this. Dont let our foolish goodtimes make you think otherwise. Thank you:)