Jieli-Tech/fw-AC63_BT_SDK #187: 最新 SDK 无法编译

mikewen, 2022-12-03:
SDK版本：               AC630N_bt_data_transfer_sdk_release_v2.2.1
芯片型号：               AC6328b
选用的cbp工程:           AC632N_spp_and_le.cbp
配置的板级:              #define CONFIG_BOARD_AC632N_DEMO
问题的分类标签：         蓝牙（BLE）
问题具体描述：           不能编译
复现的概率：             必现
复现的步骤：            编译
公版是否能复现：         Yes
提供打印输出log信息：  

||=== Build: Release in AC632N_spp_and_le (compiler: Q32S LTO Compiler) ===|
C:\JL\pi32\bin\q32s-ld.exe:..\..\..\..\cpu\bd19\liba\media_app.a|1|syntax error|
||=== Build failed: 1 error(s), 0 warning(s) (0 minute(s), 0 second(s)) ===|

JL-MinXian, 2022-12-04:
https://github.com/Jieli-Tech/fw-AC63_BT_SDK/releases/tag/ac63_bt_sdk_release_v2.2.1
[image]

mikewen, 2022-12-04:
已经是最新的 toolchain 2.4.7

.bat 也没问题，是从 github pull 的

media_app.a  是空的，是不是 media_app.a upload 错了?

-------------- Build: Release in AC632N_spp_and_le (compiler: Q32S LTO Compiler)---------------

Running target pre-build steps
clang.exe -integrated-as -g -O0 -flto -Os -Wcast-align -w -fallow-pointer-null -Wincompatible-pointer-types -Wundef -fprefer-gnu-section -Wframe-larger-than=256 -Wreturn-type -Wimplicit-function-declaration -fms-extensions -DSUPPORT_MS_EXTENSIONS -DCONFIG_RELEASE_ENABLE -DCONFIG_CPU_BD19 -DCONFIG_NEW_BREDR_ENABLE -DCONFIG_FREE_RTOS_ENABLE -DCONFIG_EQ_SUPPORT_ASYNC -DEQ_DESIGN_IN_MASK -DSINE_MAKE_IN_MASK -DCONFIG_MMU_ENABLE -DCONFIG_AEC_M=256 -DCONFIG_AUDIO_ONCHIP -DCONFIG_MEDIA_NEW_ENABLE -D__GCC_Q32S__ -DCONFIG_NEW_ECC_ENABLE -DCONFIG_CRYPTO_TOOLBOX_OSIZE_IN_MASKROM -DEVENT_HANDLER_NUM_CONFIG=2 -DEVENT_TOUCH_ENABLE_CONFIG=0 -DEVENT_POOL_SIZE_CONFIG=256 -DCONFIG_EVENT_KEY_MAP_ENABLE=0 -DTIMER_POOL_NUM_CONFIG=10 -DAPP_ASYNC_POOL_NUM_CONFIG=0 -DVFS_ENABLE=1 -DUSE_SDFILE_NEW=1 -DVFS_FILE_POOL_NUM_CONFIG=0 -DSDFILE_VERSION=0x020000 -DVM_MAX_SIZE_CONFIG=16*1024 -DVM_ITEM_MAX_NUM=128 -DCONFIG_TWS_ENABLE -DCONFIG_SPP_AND_LE_CASE_ENABLE -DCONFIG_TRANSFER_ENABLE -DCONFIG_LITE_AEC_ENABLE=0 -DCONFIG_LITE_AUDIO -DCONFIG_LMP_RESET_ENABLE -DCONFIG_LMP_REFRESH_ENCRYPTION_KEY_ENABLE -DCONFIG_LINK_DISTURB_SCAN_ENABLE=0 -DCONFIG_BTCTRLER_TASK_DEL_ENABLE -DCONFIG_UPDATA_ENABLE -DCONFIG_OTA_UPDATA_ENABLE -DCONFIG_ITEM_FORMAT_VM -DTCFG_APP_BT_EN=1 -DUSB_PC_NO_APP_MODE -D__LD__ -flto -target q32s -integrated-as -fno-builtin -mllvm -inline-threshold=5 -Oz -I..\..\..\..\include_lib -I..\..\..\..\include_lib\driver -I..\..\..\..\include_lib\driver\device -I..\..\..\..\include_lib\driver\cpu\bd19 -I..\..\..\..\include_lib\system -I..\..\..\..\include_lib\system\generic -I..\..\..\..\include_lib\system\device -I..\..\..\..\include_lib\system\fs -I..\..\..\..\include_lib\system\ui -I..\..\..\..\include_lib\btctrler -I..\..\..\..\include_lib\btctrler\port\bd19 -I..\..\..\..\include_lib\update -I..\..\..\..\include_lib\agreement -I..\..\..\..\include_lib\btstack\third_party\common -I..\..\..\..\include_lib\btstack\third_party\rcsp -I..\..\..\..\include_lib\media\media_new -I..\..\..\..\include_lib\media\media_new\media -I..\..\..\..\include_lib\media\media_new\media\cpu\bd19 -I..\..\..\..\include_lib\media\media_new\media\cpu\bd19\asm -I..\..\..\..\apps\spp_and_le\include -I..\..\..\..\apps\spp_and_le\board\bd19 -I..\..\..\..\apps\common -I..\..\..\..\apps\common\device -I..\..\..\..\apps\common\include -I..\..\..\..\apps\common\power_manage -I..\..\..\..\cpu\bd19 -I..\..\..\..\apps\common\third_party_profile\jieli -I..\..\..\..\apps\common\third_party_profile\jieli\JL_rcsp -I..\..\..\..\apps\common\third_party_profile\jieli\online_db -I..\..\..\..\apps\common\third_party_profile\common -I..\..\..\..\include_lib\ai_stack\JL_rcsp -I..\..\..\..\include_lib\btstack -I..\..\..\..\apps\common\third_party_profile\jieli\JL_rcsp\bt_trans_data -I..\..\..\..\apps\common\third_party_profile\jieli\JL_rcsp\rcsp_updata -I..\..\..\..\apps\common\third_party_profile\jieli\gatt_common -I..\..\..\..\apps\common\third_party_profile\Tecent_LL\include -I..\..\..\..\apps\common\third_party_profile\Tecent_LL\tecent_ll_demo -I..\..\..\..\apps\spp_and_le\examples\ll_sync -I..\..\..\..\apps\common\cJSON -I..\..\..\..\apps\common\third_party_profile\tuya_protocol -I..\..\..\..\apps\common\third_party_profile\tuya_protocol\app\demo -I..\..\..\..\apps\common\third_party_profile\tuya_protocol\app\product_test -I..\..\..\..\apps\common\third_party_profile\tuya_protocol\app\uart_common -I..\..\..\..\apps\common\third_party_profile\tuya_protocol\extern_components\mbedtls -I..\..\..\..\apps\common\third_party_profile\tuya_protocol\port -I..\..\..\..\apps\common\third_party_profile\tuya_protocol\sdk\include -I..\..\..\..\apps\common\third_party_profile\tuya_protocol\sdk\lib -I..\..\..\..\apps\spp_and_le\examples\tuya -I..\..\..\..\apps\common\third_party_profile\hilink_protocol -I..\..\..\..\apps\common\third_party_profile\hilink_protocol\mbedtls_protocol -I..\..\..\..\apps\common\third_party_profile\hilink_protocol\mbedtls_protocol\mbedtls -I..\..\..\..\apps\common\third_party_profile\hilink_protocol\mbedtls_protocol\mbedtls\configs -I..\..\..\..\apps\common\third_party_profile\hilink_protocol\mbedtls_protocol\mbedtls\psa -I..\..\..\..\apps\spp_and_le\examples\hilink -I..\..\..\..\cpu\bd19\audio_encode -I..\..\..\..\cpu\bd19\audio_decode -I..\..\..\..\apps\common\audio -I..\..\..\..\apps\common\device\usb -I..\..\..\..\apps\common\device\usb\device -I..\..\..\..\apps\common\device\usb\host -IC:\JL\pi32\q32s-include -E -P ..\..\..\..\cpu\bd19\sdk_used_list.c -o ..\..\..\..\cpu\bd19\sdk_used_list.used
clang.exe -integrated-as -g -O0 -flto -Os -Wcast-align -w -fallow-pointer-null -Wincompatible-pointer-types -Wundef -fprefer-gnu-section -Wframe-larger-than=256 -Wreturn-type -Wimplicit-function-declaration -fms-extensions -DSUPPORT_MS_EXTENSIONS -DCONFIG_RELEASE_ENABLE -DCONFIG_CPU_BD19 -DCONFIG_NEW_BREDR_ENABLE -DCONFIG_FREE_RTOS_ENABLE -DCONFIG_EQ_SUPPORT_ASYNC -DEQ_DESIGN_IN_MASK -DSINE_MAKE_IN_MASK -DCONFIG_MMU_ENABLE -DCONFIG_AEC_M=256 -DCONFIG_AUDIO_ONCHIP -DCONFIG_MEDIA_NEW_ENABLE -D__GCC_Q32S__ -DCONFIG_NEW_ECC_ENABLE -DCONFIG_CRYPTO_TOOLBOX_OSIZE_IN_MASKROM -DEVENT_HANDLER_NUM_CONFIG=2 -DEVENT_TOUCH_ENABLE_CONFIG=0 -DEVENT_POOL_SIZE_CONFIG=256 -DCONFIG_EVENT_KEY_MAP_ENABLE=0 -DTIMER_POOL_NUM_CONFIG=10 -DAPP_ASYNC_POOL_NUM_CONFIG=0 -DV [... cut ...] /../../cpu/bd19/tools/sdk.map -flto --plugin-opt=-inline-threshold=5  C:\JL\pi32\q32s-lib\libm.a C:\JL\pi32\q32s-lib\libc.a C:\JL\pi32\q32s-lib\libm.a C:\JL\pi32\q32s-lib\libcompiler-rt.a
C:\JL\pi32\bin\q32s-ld.exe:../../../../cpu/bd19/liba/media_app.a: file format not recognized; treating as linker script
C:\JL\pi32\bin\q32s-ld.exe:../../../../cpu/bd19/liba/media_app.a:1: syntax error
Process terminated with status 1 (0 minute(s), 0 second(s))
1 error(s), 0 warning(s) (0 minute(s), 0 second(s))

JL-MinXian, 2022-12-05:
你是怎么下载程序包测试？这边试过是可以编译通过，bd19的cpu目录下有media_app.a文件；下载方式如下
[image]