Jieli-Tech/fw-AC63_BT_SDK #189: [REQUEST] Mr. Wanbaolu

zonalimitatore, 2022-12-11:
Hello. I've buy an item called "Vgate BLE".
Now, I've discovered that in the description has been called BM2 software. But in reality, work only with an app (not available on GooglePlay).
The app name is "Battery Asst" (found only scanning the qrcode on the item). Forward to a strange item and download it.

Well, what the app do, is nothing about was described. Maybe is not your fault, but the programmer (wanbaolu) not reply to our mails.
The app isn't usable, it's bugged, and not report the history. 
I write here, just because, Mr. wanbaolu has done apps for your chip (only 12 on PlayStore). I can't figure it out how many outside...
And maybe, this can help he to reply.

I don't ask too much, just an app that do what was described for.  
Because actually it's just a "Real Time Vmeter". No history, no alarms, no notify, no update, encrypted source code, no way to find "com.ifly.battery" folder on phone memory (even in debug mode) for search historical.

To concluding... We're already in 84 people with the same problems. Before we do a class action, i've preferred to write here, for the reasons described before.
So, i really hope and appreciate a contact from Mr. wanbaolu.
Thank you.

0x0fe, 2023-02-25:
"We're already in 84 people with the same problems. Before we do a class action, i've preferred to write here," )
pathetically comic, Your rambling nonsense has nothing to do with Jieli.

zonalimitatore, 2023-02-26:
And who haved said that? We say "contact from wanbaolu"... And he never contact us. But, hey, it's ok!We live in a free world.Btw from 84 people now are 112.

0x0fe, 2023-02-26:
are you for real man? threatening JL to "do a class action" because a random APP doesnt work as you expect, in a github issue ticket of a Zhuhai SoC company which has nothign to do in the matter? 
Get real, you must be living in your own parallel universe... Anyway, this is an issue tracker for engineers using JL chips in their design (us), not a place for delusional nonsense.
If you have a complain abouta random 3usd app, contact the editor, and even then you will learn that you cannot "do a class action" against anyone because your 3sud app does not do what you though it would do. That is so ridiculous, i hope you are very young otherwise this is concerning.

zonalimitatore, 2023-02-26:
No for real. You mean that "you" don't know who is wanbaolu? Really? 
And no, we're not young.
We know wanbaolu is here, with other nick, so please don't play with us. And maybe "you" don't know OBD2 seller on AliExpress, or vgate reseller?
When someone buyed a thing, is expected that work and do what seller say. And not a fully not working stuff