Jieli-Tech/fw-AC63_BT_SDK #194: ADC问题: ADC 时钟频率

mikewen, 2023-01-27:
AC632N_datasheet 中有 ADC 寄存器说明. 
其他 CPU 系类  ( AC636N ) 的ADC 寄存器是一样的吗?

LSB 时钟 通过 clk_get("lsb")获取，LSB 时钟 不同系列最高不同?
AC6328A 最高 96/2 = 48 MHz ? 
AC6369F 最高 160/2 = 80 MHz ? 
其时钟最大不可超过 1MHz, 那如果 ADC_BAUD 用 12 分频 = 48/12 = 4MHz,  会怎样？ADC 不准吗?

lawrencejiabin, 2023-01-28:
[AC636X用户手册开源版本V1.1.pdf](https://github.com/Jieli-Tech/fw-AC63_BT_SDK/files/10524231/AC636X.V1.1.pdf)
可以参考这个说明，只能到1M而已
https://gitee.com/jieli-tech_fae/IOT-AC630N/blob/master/doc/software/驱动/adc.md

mikewen, 2023-01-28:
非常感谢!  AC636X用户手册 太管用了!  

手册里的 序言 是AC696X 系列芯片，AC696X 和 AC636X 的寄存器是一样的？

ACCount12, 2023-03-17:
AC696X 和 AC636X 使用相同的硅 (BR25)。 主要区别在于 AC696X 启用了模拟音频支持。 它具有专用音频 ADC/DAC 和 FM RX/TX。

其他寄存器应该匹配。 我不确定数字音频外围设备（Audio IIS、PDM Link）是否可以在 AC636X 中工作，但我不明白为什么它们不能。 您询问的通用 ADC 应该以相同的方式工作。