Jieli-Tech/fw-AC63_BT_SDK #202: [AC69xx] SDK repository location

hungnvk54, 2023-11-03:
Dear developer,

I've read the description of this project. It say that ```Firmware Generic Bluetooth SDK（AC63 series）, Support AC631N/AC635N/AC636N/AC637N/AC632N, compatible with AC69 series without audio support``` but I can't found the AC69 Serial SDK.

Can you help me to find the AC69 Serial SDK?

Thank you for reading.

JL-MinXian, 2023-11-03:
Using the same sdk, refer to the instructions：
https://gitee.com/Jieli-Tech/fw-AC63_BT_SDK/issues/I4Y5CG

hungnvk54, 2023-11-03:
> Using the same sdk, refer to the instructions： https://gitee.com/Jieli-Tech/fw-AC63_BT_SDK/issues/I4Y5CG

Woa, thank for your responding.

Sorry, I don't provide more detail. I've developed with AC6905A. I have read your instructions but didn't find the equivalent. Can you give me more information to develop AC6905A?

Thank you so much.

JL-MinXian, 2024-02-18:
Sorry, AC63 SDK not support  AC6905A.