Jieli-Tech/fw-AC63_BT_SDK #211: Documenting setup process on arch linux

Kanken6174, 2025-03-21:
Hi, this isn't really an issue (though i'm sure this will be somewhat difficult), it's just me documenting how i'm setting this up so others can do it more easily on an arch linux machine.

# Downloading ressources
Grab these:
[toolchain](https://pkgman.jieliapp.com/s/linux-toolchain)
[post process and upload tools](https://pkgman.jieliapp.com/s/linux-postbuild)

These should be decompressed in `/opt/jieli`. You're typically expecting this end result:
```bash
➜  / tree -L2 /opt/jieli
/opt/jieli
├── README.md
├── common
│   ├── bin
│   ├── include
│   └── lib
├── pi32
│   ├── bin
│   ├── include
│   └── lib
├── pi32v2
│   ├── bin
│   ├── include
│   └── lib
├── postbuild
│   ├── fat_comm
│   ├── fw_add
│   ├── isd_download
│   ├── json_to_res
│   ├── packres
│   ├── remove_tailing_zeros
│   └── ufw_maker
└── q32s
    ├── bin
    ├── include
    └── lib
```

You need to create the postbuild folder yourself and put the post processing tools in there. Don't forget to add the postbuilt tools folder to your PATH variable.

now you'll need to clone this repo wherever you want, i usuall do that in `~/code` with `git clone https://github.com/Jieli-Tech/fw-AC63_BT_SDK.git` (obvious)

# compiling
 
compiling is done with `make` at the root of the repo, if you've done the previous step right it should work right until after the linking step:

```bash
+CC ../../../../cpu/br34/tone_player.c
+CC ../../../../cpu/br34/uart_dev.c
+LINK ../../../../cpu/br34/tools/sdk.elf
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (536) in hci_send_event
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (276) in sdp_send_service_search_attribute_request
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (328) in att_handle_request
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (360) in sdp_handle_service_search_request
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (356) in sdp_handle_service_attribute_request
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (376) in sdp_handle_service_search_attribute_request
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (260) in EccPoint_mult
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (436) in hmacCompute
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (320) in sha256Compute
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (320) in btrx_dctrim
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (364) in btrx_dc_compensate
/opt/jieli/pi32v2/bin/ld: warning: LLVM gold plugin: stack size limit exceeded (540) in hci_send_event_cmd_complete
+POST-BUILD
/opt/jieli/pi32v2/bin/objdump: 'sdk.elf': No such file or directory
/opt/jieli/pi32v2/bin/objcopy: sdk.elf not found/opt/jieli/pi32v2/bin/objcopy: sdk.elf not found/opt/jieli/pi32v2/bin/objcopy: sdk.elf not found/opt/jieli/pi32v2/bin/objcopy: sdk.elf not found/opt/jieli/pi32v2/bin/objcopy: sdk.elf not found/opt/jieli/pi32v2/bin/objcopy: sdk.elf not found/opt/jieli/pi32v2/bin/objcopy: sdk.elf not found/opt/jieli/pi32v2/bin/objcopy: sdk.elf not found
../../../../cpu/br34/tools/download.sh: line 45: lz4_packet: command not found
/opt/jieli/pi32v2/bin/objdump: 'sdk.elf': No such file or directory
/opt/jieli/pi32v2/bin/objsizedump: 'sdk.elf': No such file or directory
cat: text.bin: No such file or directory
cat: data.bin: No such file or directory
cat: data_code.bin: No such file or directory
cat: bank.bin: No such file or directory
../../../../cpu/br34/tools/download.sh: line 52: /opt/utils/strip-ini: No such file or directory
../../../../cpu/br34/tools/download.sh: line 58: host-client: command not found
make[1]: *** [Makefile:570: all] Error 127
make[1]: Leaving directory '/home/kanken/code/fw-AC63_BT_SDK/apps/spp_and_le/board/br34'
make: *** [Makefile:37: ac638n_spp_and_le] Error 2
➜  fw-AC63_BT_SDK git:(master) ✗ 
```

The post processing code seems to mess up here, i'm still in the process of trying to figure it out.

Kanken6174, 2025-03-21:
The errors that happen after the linking step are caused by a file called `download.sh` that is most likely responsible for uploading the binary to the micro controller. This file seems auto-generated in a very strange way and the original is from `<project_root>/cpu/<cpu type here, for me it's bd19 since i'm using an AC6329C4>/tools/download.c`

This file appears to have trouble with paths. The .elf does exist, it's generated by the make command right there: `<project_root>/cpu/bd19/tools/sdk.elf`

To keep things clean to debug, i'm only `make`-ing the project in `/home/kanken/code/fw-AC63_BT_SDK/apps/hid/board/bd19`, that should keep the duplicated files to a minimum while i figure this out

Kanken6174, 2025-03-21:
In my case i had to select this config for my chip, but it's not the main issue:
```c
/*
 *Board level configuration selection
 */

//#define CONFIG_BOARD_AC632N_DEMO
// #define CONFIG_BOARD_AC6321A_DEMO
// #define CONFIG_BOARD_AC6321A_STAND_KEYBOARD//AC6321A+AD15键盘项目,download.bat需要修改添加打包ex_mcu.bin
// #define CONFIG_BOARD_AC6321A_MOUSE
// #define CONFIG_BOARD_AC6323A_DEMO
// #define CONFIG_BOARD_AC6328A_KEYFOB
// #define CONFIG_BOARD_AC6328B_DEMO
// #define CONFIG_BOARD_AC6329B_DEMO
#define CONFIG_BOARD_AC6329C_DEMO
// #define CONFIG_BOARD_AC6329E_DEMO
// #define CONFIG_BOARD_AC6329F_DEMO
```
in `fw-AC63_BT_SDK/apps/hid/board/bd19/board_config.h`

Kanken6174, 2025-03-21:
for now i'll hardcode the path:
```Make
all: pre_build $(OUT_ELF)
	$(info +POST-BUILD)
	$(QUITE) $(RUN_POST_SCRIPT) ../../../../cpu/bd19/tools/sdk
```

This seems to help somewhat:
```log
+POST-BUILD

../../../../cpu/bd19/tools/download.sh: line 64: lz4_packet: command not found

../../../../cpu/bd19/tools/sdk.elf:     file format ELF32-q32s

Sections:
Idx Name          Size      Address          Type
  0               00000000 0000000000000000 
  1 .text         00033f98 0000000001e00120 TEXT DATA 
  2 .mmu_tlb      00000800 0000000000000000 DATA 
  3 .boot_info    00000040 0000000000000800 DATA 
  4 .data         00000960 0000000000000840 DATA 
  5 .irq_stack    00000820 00000000000011a0 DATA 
  6 .bss          00003ec8 00000000000019c0 BSS
  7 .data_code    0000274c 00000000000058a0 TEXT DATA 
  8 .nv_bss       00000344 0000000000008000 DATA 
  9 .overlay_aec  00000000 0000000000008344 DATA 
 10 .overlay_aac  00000000 0000000000008344 DATA 
 11 .ram0_empty0  00000004 0000000000008344 BSS
 12 .overlay_aec_ram 00000000 0000000000008348 DATA 
 13 .overlay_aac_ram 00000000 0000000000008348 DATA 
 14 .debug_str    0003d256 0000000000000000 
 15 .debug_loc    00021429 0000000000000000 
 16 .debug_abbrev 00001016 0000000000000000 
 17 .debug_info   000a79a6 0000000000000000 
 18 .debug_ranges 00005120 0000000000000000 
 19 .debug_macinfo 00000001 0000000000000000 
 20 .debug_pubnames 0001e476 0000000000000000 
 21 .debug_pubtypes 00010957 0000000000000000 
 22 .debug_frame  000098d0 0000000000000000 
 23 .debug_line   00042e46 0000000000000000 
 24 .debug_aranges 00000048 0000000000000000 
 25 .shstrtab     0000013c 0000000000000000 
 26 .symtab       00098870 0000000000000000 
 27 .strtab       0000d8f5 0000000000000000 
cat: bank.bin: No such file or directory
../../../../cpu/bd19/tools/download.sh: line 71: /opt/utils/strip-ini: No such file or directory
../../../../cpu/bd19/tools/download.sh: line 77: host-client: command not found
make: *** [Makefile:425: all] Error 127
```

but it seems to be trying to call `lz4_packet` which isn't the right name in my linux for `lz4`... after some research `lz4_packet ` appears to be a custom app from jieli, it's right there:
```bash
➜  fw-AC63_BT_SDK git:(master) ✗ find . -name "lz4_packet" 
./cpu/br23/tools/lz4_packet
./cpu/bd29/tools/lz4_packet
./cpu/br25/tools/lz4_packet
./cpu/br34/tools/lz4_packet
./cpu/bd19/tools/lz4_packet
./cpu/br30/tools/lz4_packet
```
I'll add `./cpu/bd19/tools/` to my path...

Kanken6174, 2025-03-21:
this is required it seems:
```c
chmod +x <your project root here>/cpu/bd19/tools/* 
```

Kanken6174, 2025-03-21:
`strip-ini` does not seem to be anywhere at all... hmm

Kanken6174, 2025-03-21:
looking at `download.bat` i can theorize on how to use isd_download to flash the device, but it won't be that easy:
```bat
@echo off

cd %~dp0

copy ..\..\tone.cfg .
copy ..\..\cfg_tool.bin .
copy ..\..\app.bin .
copy ..\..\bd19loader.bin .
copy ..\..\p11_code.bin .
copy ..\..\script.ver .
copy ..\..\flash_params.bin

..\..\isd_download.exe ..\..\isd_config.ini -tonorflash -dev bd19 -boot 0x2000 -div8 -wait 300 -uboot ..\..\uboot.boot -app ..\..\app.bin ..\..\cfg_tool.bin -res ..\..\p11_code.bin -uboot_compress -flash-params flash_params.bin
:: -format all
::-reboot 2500

@rem ɾ����ʱ�ļ�-format all
if exist *.mp3 del *.mp3 
if exist *.PIX del *.PIX
if exist *.TAB del *.TAB
if exist *.res del *.res
if exist *.sty del *.sty

@rem generate upgrade file
..\..\fw_add.exe -noenc -fw jl_isd.fw -add ..\..\ota.bin -type 100 -out jl_isd.fw
@rem add ver to jl_isd.fw
..\..\fw_add.exe -noenc -fw jl_isd.fw -add script.ver -out jl_isd.fw

..\..\ufw_maker.exe -fw_to_ufw jl_isd.fw
copy jl_isd.ufw update.ufw
del jl_isd.ufw

@rem ��������˵��
@rem -format vm        //����VM ����
@rem -format cfg       //����BT CFG ����
@rem -format 0x3f0-2   //��ʾ�ӵ� 0x3f0 �� sector ��ʼ�������� 2 �� sector(��һ������Ϊ16���ƻ�10���ƶ��ɣ��ڶ�������������10����)

ping /n 2 127.1>null
IF EXIST null del null
```

Kanken6174, 2025-03-21:
I modified download.c quite heavily, now it looks like this:
```c
// *INDENT-OFF*
#include "app_config.h"

#ifdef __SHELL__

##!/bin/sh

${OBJDUMP} -D -address-mask=0x1ffffff -print-dbg ../$2/$1.elf > ../$2/$1.lst
${OBJCOPY} -O binary -j .text ../$2/$1.elf text.bin
${OBJCOPY} -O binary -j .data ../$2/$1.elf data.bin
${OBJCOPY} -O binary -j .data_code ../$2/$1.elf data_code.bin
${OBJCOPY} -O binary -j .overlay_aec ../$2/$1.elf aec.bin
${OBJCOPY} -O binary -j .overlay_aac ../$2/$1.elf aac.bin
${OBJCOPY} -O binary -j .overlay_aptx ../$2/$1.elf aptx.bin

${OBJCOPY} -O binary -j .common ../$2/$1.elf common.bin

bank_files=
for i in $(seq 0 20)
do
    ${OBJCOPY} -O binary -j .overlay_bank$i ../$2/$1.elf bank$i.bin
    if [ ! -s bank$i.bin ]
    then
        break
    fi
    bank_files=$bank_files"bank$i.bin 0x0 "
done
echo $bank_files
lz4_packet -dict text.bin -input common.bin 0 $bank_files -o bank.bin

${OBJDUMP} -section-headers -address-mask=0x1ffffff ../$2/$1.elf
${OBJSIZEDUMP} -lite -skip-zero -enable-dbg-info ../$2/$1.elf | sort -k 1 > symbol_tbl.txt

cat text.bin data.bin data_code.bin aec.bin aac.bin bank.bin aptx.bin > app.bin

/*/opt/utils/strip-ini -i isd_config.ini -o isd_config.ini*/
cd /home/kanken/code/fw-AC63_BT_SDK/cpu/bd19/tools/
/* files="app.bin ${CPU}loader.* uboot*  ota*.bin p11_code.bin isd_config.ini isd_download.exe fw_add.exe ufw_maker.exe" */
files="app.bin ${CPU}loader.* uboot*  ota*.bin p11_code.bin isd_config.ini flash_params.bin"
NICKNAME="${CPU}_sdk"
pwd
//host-client -project ${NICKNAME}$2 -f ${files} $1.elf
isd_download $2/isd_config.ini -tonorflash -dev bd19 -boot 0x2000 \
 -div8 -wait 300 -uboot $2/uboot.boot -app $3/app.bin $2/cfg_tool.bin \
 -res $2/p11_code.bin -uboot_compress -flash-params flash_params.bin

cd -;
#else
///...
```
And is called like this:
```Make
all: pre_build $(OUT_ELF)
	$(info +POST-BUILD)
	$(QUITE) $(RUN_POST_SCRIPT) sdk ../../../cpu/bd19/tools ../../../apps/hid/board/bd19
```

Current result:
```log
/home/kanken/code/fw-AC63_BT_SDK/cpu/bd19/tools
INI PATH: /home/kanken/code/fw-AC63_BT_SDK/cpu/bd19/tools/isd_config.ini
Device offline, only package the file.
SPI nor flash online
ota.bin: /home/kanken/code/fw-AC63_BT_SDK/cpu/bd19/tools/ota.bin
--------------------------- OTA UPDATE INFO ---------------------------
| VM size = 0x3000
| !!!!!! FAIL: testbox BLE upgrade(size=0xbc38) require minial space size 0xc000
        ^^^^^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| !!!!!! FAIL: testbox classic Bluetooth upgrade(size=0x9abc) require minial space size 0xa000
        ^^^^^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| !!!!!! FAIL: BLE rcsp upgrade(size=0x10640) require minial space size 0x11000
        ^^^^^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| !!!!!! FAIL: custom UART upgrade(size=0x4090) require minial space size 0x5000
        ^^^^^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| !!!!!! FAIL: USB hid upgrade(size=0x6514) require minial space size 0x7000
        ^^^^^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| no valid ota upgrade found               
-----------------------------------------------------------------------
--------------------FLASH INFO--------------------
|  PID : AC632N                                  |
|  VID : 0.01                                    |
|  FLASH_BIN_SIZE : 0x3b000                      |
|  FLASH_NEED_SIZE : 0x3f000                     |
|  FLASH_REAL_SIZE : UNLIMITED                   |
|  VM_REAL_SIZE : 0x3000                         |
|  VM_START_ADDR : 0x3b000                       |
|  VM_END_ADDR : 0x3e000                         |
|  BTIF_RESERVED_SIZE : 0x1000                   |
|  BTIF_RESERVED_START : 0x3e000                 |
|  BTIF_RESERVED_END : 0x3f000                   |
|  LAVE_SIZE : 0x1000                            |
|  ENTRY_ADDR : 0x1e00120                        |
|  ERASE MODE : NONE                             |
|  UBOOT VERSION : 1970-01-01 01:00:00           |
--------------------------------------------------
ISDdownload
/home/kanken/code/fw-AC63_BT_SDK/apps/hid/board/bd19
```

Kanken6174, 2025-03-21:
It seems to at least recognize my chip which is a good sign?

Kanken6174, 2025-03-21:
Checking on #167 ... is it true? This sdk does not support downloading to the mcu on linux? Then what's the point of Jieli packaging all the post-processing tools for linux? it wouldn't make sense!

Kanken6174, 2025-03-21:
from https://doc.zh-jieli.com/Tools/zh-cn/dev_tools/dev_env/index.html#linux :
> 1.2. Linux environment

>By default, the SDK project supports development and compilation using Makefile in Linux environment. Please refer to[ Compiling with Make ](https://doc.zh-jieli.com/Tools/zh-cn/dev_tools/build_download/build_with_makefile.html).

    SDK post-processing script download.shAfter modification (refer to the corresponding Windows version download.bat) can be generated .fwFirmware files.

    **Downloading firmware to the device is not supported.**

Wow that really sucks... I don't use windows for anything so i guess this chip series is in a dead end for me?

ACCount12, 2025-04-02:
There are unofficial chip burning tools that may or may not work for you.
Want to try?

Kanken6174, 2025-04-14:
> There are unofficial chip burning tools that may or may not work for you. Want to try?

I tried jl-uboot-tool already, it doesn't work with the chips i have on hand, and with LCSC delisting all Jieli chips (of which they were the sole available supplier), i'm quickly loosing interest in them, as even if i get this mess of a tool-chain to work end-to-end, i wouldn't be able to use these chips at scale anyways.